orocos/orocos_kinematics_dynamics #296: make error --python_orocos_kdl

Mr-Xia-6, 2020-11-08:
orocos_kinematics_dynamics/python_orocos_kdl/build$ make -j8
[ 10%] Building CXX object CMakeFiles/python_module_PyKDL.dir/PyKDL/sipPyKDLpart1.cpp.o
[ 20%] Building CXX object CMakeFiles/python_module_PyKDL.dir/PyKDL/sipPyKDLpart2.cpp.o
[ 30%] Building CXX object CMakeFiles/python_module_PyKDL.dir/PyKDL/sipPyKDLpart3.cpp.o
[ 40%] Building CXX object CMakeFiles/python_module_PyKDL.dir/PyKDL/sipPyKDLpart5.cpp.o
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart2.cpp: In member function ‘void sipChainJntToJacSolver::updateInternalDataStructures()’:
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart2.cpp:76:10: error: ‘updateInternalDataStructures’ is not a member of ‘KDL::ChainJntToJacSolver’
          ::ChainJntToJacSolver::updateInternalDataStructures();
          ^
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart2.cpp: In function ‘PyObject* meth_ChainJntToJacSolver_updateInternalDataStructures(PyObject*, PyObject*)’:
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart2.cpp:135:62: error: ‘class KDL::ChainJntToJacSolver’ has no member named ‘updateInternalDataStructures’
             (sipSelfWasArg ? sipCpp-> ::ChainJntToJacSolver::updateInternalDataStructures() : sipCpp->updateInternalDataStructures());
                                                              ^
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart2.cpp:135:103: error: ‘class KDL::ChainJntToJacSolver’ has no member named ‘updateInternalDataStructures’
             (sipSelfWasArg ? sipCpp-> ::ChainJntToJacSolver::updateInternalDataStructures() : sipCpp->updateInternalDataStructures());
                                                                                                       ^
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp: In member function ‘void sipChainIdSolver_RNE::updateInternalDataStructures()’:
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp:1997:10: error: ‘updateInternalDataStructures’ is not a member of ‘KDL::ChainIdSolver_RNE’
 extern "C" {static PyObject *meth_ChainIdSolver_RNE_CartToJnt(PyObject *, PyObj
          ^
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp: In function ‘PyObject* meth_ChainIdSolver_RNE_updateInternalDataStructures(PyObject*, PyObject*)’:
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp:2078:60: error: ‘class KDL::ChainIdSolver_RNE’ has no member named ‘updateInternalDataStructures’
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp:2078:101: error: ‘class KDL::ChainIdSolver_RNE’ has no member named ‘updateInternalDataStructures’
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp: In function ‘void* cast_ChainIdSolver_RNE(void*, const sipTypeDef*)’:
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp:2102:31: error: ‘SolverI’ in namespace ‘::’ does not name a type
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp:2102:39: error: expected ‘>’ before ‘*’ token
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp:2102:39: error: expected ‘(’ before ‘*’ token
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp:2102:40: error: expected primary-expression before ‘>’ token
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp:2102:49: error: expected ‘)’ before ‘;’ token
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp: In function ‘void* cast_ChainIdSolver(void*, const sipTypeDef*)’:
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp:2265:31: error: ‘SolverI’ in namespace ‘::’ does not name a type
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp:2265:39: error: expected ‘>’ before ‘*’ token
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp:2265:39: error: expected ‘(’ before ‘*’ token
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp:2265:40: error: expected primary-expression before ‘>’ token
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart1.cpp:2265:49: error: expected ‘)’ before ‘;’ token
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart3.cpp: In member function ‘void sipChainIkSolverPos_NR_JL::updateInternalDataStructures()’:
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart3.cpp:87:10: error: ‘updateInternalDataStructures’ is not a member of ‘KDL::ChainIkSolverPos_NR_JL’
          ::ChainIkSolverPos_NR_JL::updateInternalDataStructures();
          ^
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart3.cpp: In function ‘PyObject* meth_ChainIkSolverPos_NR_JL_updateInternalDataStructures(PyObject*, PyObject*)’:
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart3.cpp:162:65: error: ‘class KDL::ChainIkSolverPos_NR_JL’ has no member named ‘updateInternalDataStructures’
             (sipSelfWasArg ? sipCpp-> ::ChainIkSolverPos_NR_JL::updateInternalDataStructures() : sipCpp->updateInternalDataStructures());
                                                                 ^
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart3.cpp:162:106: error: ‘class KDL::ChainIkSolverPos_NR_JL’ has no member named ‘updateInternalDataStructures’
             (sipSelfWasArg ? sipCpp-> ::ChainIkSolverPos_NR_JL::updateInternalDataStructures() : sipCpp->updateInternalDataStructures());
                                                                                                          ^
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart5.cpp: In function ‘void* init_type_Segment(sipSimpleWrapper*, PyObject*, PyObject*, PyObject**, PyObject**, PyObject**)’:
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart5.cpp:921:39: error: ‘Fixed’ is not a member of ‘KDL::Joint’
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart5.cpp:945:39: error: ‘Fixed’ is not a member of ‘KDL::Joint’
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart3.cpp: In member function ‘void sipChainIkSolverPos_NR::updateInternalDataStructures()’:
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart3.cpp:410:10: error: ‘updateInternalDataStructures’ is not a member of ‘KDL::ChainIkSolverPos_NR’
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart3.cpp: In function ‘PyObject* meth_ChainIkSolverPos_NR_updateInternalDataStructures(PyObject*, PyObject*)’:
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart3.cpp:485:62: error: ‘class KDL::ChainIkSolverPos_NR’ has no member named ‘updateInternalDataStructures’
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart3.cpp:485:103: error: ‘class KDL::ChainIkSolverPos_NR’ has no member named ‘updateInternalDataStructures’
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart3.cpp: In function ‘PyObject* meth_ChainIkSolverVel_updateInternalDataStructures(PyObject*, PyObject*)’:
[... cut ...]
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build/PyKDL/sipPyKDLpart2.cpp:2068:105: error: ‘class KDL::ChainIkSolverVel_pinv’ has no member named ‘updateInternalDataStructures’
CMakeFiles/python_module_PyKDL.dir/build.make:134: recipe for target 'CMakeFiles/python_module_PyKDL.dir/PyKDL/sipPyKDLpart2.cpp.o' failed
make[2]: *** [CMakeFiles/python_module_PyKDL.dir/PyKDL/sipPyKDLpart2.cpp.o] Error 1
CMakeFiles/Makefile2:141: recipe for target 'CMakeFiles/python_module_PyKDL.dir/all' failed
make[1]: *** [CMakeFiles/python_module_PyKDL.dir/all] Error 2
Makefile:159: recipe for target 'all' failed
make: *** [all] Error 2

MatthijsBurgh, 2020-11-08:
Since #294, PyKDL by SIP is removed. So only the PyKDL version by PyBind is available. So please use that version. Follow the install instructions [here](https://github.com/orocos/orocos_kinematics_dynamics/blob/master/python_orocos_kdl/INSTALL.md).

P.S. Next time you have an issue, please more information than only the make output. Such as some context, systems information, git branch/hash/tag etc.

Mr-Xia-6, 2020-11-09:
> Since #294, PyKDL by SIP is removed. So only the PyKDL version by PyBind is available. So please use that version. Follow the install instructions [...]

THANKS

I test it in ubuntu16.04, however when I try to make, I faced the following error:

OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/build$ make[ 16%] Building CXX object CMakeFiles/PyKDL.dir/PyKDL/frames.cpp.o
In file included from /home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/PyKDL.h:25:0,
                 from /home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:27:
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h: In instantiation of ‘void pybind11::cpp_function::initialize(Func&&, Return (*)(Args ...), const Extra& ...) [with Func = pybind11::cpp_function::cpp_function(Return (Class::*)(Arg ...) const, const Extra& ...) [with Return = double; Class = KDL::Vector; Arg = {}; Extra = {pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::arg_v}]::<lambda(const KDL::Vector*)>; Return = double; Args = {const KDL::Vector*}; Extra = {pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::arg_v}]’:
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:100:9:   required from ‘pybind11::cpp_function::cpp_function(Return (Class::*)(Arg ...) const, const Extra& ...) [with Return = double; Class = KDL::Vector; Arg = {}; Extra = {pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::arg_v}]’
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:1267:73:   required from ‘pybind11::class_<type_, options>& pybind11::class_<type_, options>::def(const char*, Func&&, const Extra& ...) [with Func = double (KDL::Vector::*)() const; Extra = {pybind11::arg_v}; type_ = KDL::Vector; options = {}]’
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:92:61:   required from here
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:157:9: error: static assertion failed: The number of argument annotations does not match the number of function arguments
         static_assert(expected_num_args<Extra...>(sizeof...(Args), cast_in::has_args, cast_in::has_kwargs),
         ^
/home/yusheng/xzq_ws/src/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:161:19: error: cannot convert ‘pybind11::cpp_function::initialize(Func&&, Return (*)(Args ...), const Extra& ...) [with Func = pybind11::cpp_function::cpp_function(Return (Class::*)(Arg ...) const, const Extra& ...) [with Return = double; Class = KDL::Vector; Arg = {}; Extra = {pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::arg_v}]::<lambda(const KDL::Vector*)>; Return = double; Args = {const KDL::Vector*}; Extra = {pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::arg_v}]::<lambda(pybind11::detail::function_call&)>’ to ‘pybind11::handle (*)(pybind11::detail::function_call&)’ in assignment
         rec->impl = [](function_call &call) -> handle {
                   ^
CMakeFiles/PyKDL.dir/build.make:94: recipe for target 'CMakeFiles/PyKDL.dir/PyKDL/frames.cpp.o' failed
make[2]: *** [CMakeFiles/PyKDL.dir/PyKDL/frames.cpp.o] Error 1
CMakeFiles/Makefile2:324: recipe for target 'CMakeFiles/PyKDL.dir/all' failed
make[1]: *** [CMakeFiles/PyKDL.dir/all] Error 2
Makefile:159: recipe for target 'all' failed
make: *** [all] Error 2

MatthijsBurgh, 2020-11-09:
@Mr-Xia-6 are you using a catkin workspace?

Mr-Xia-6, 2020-11-09:
> @Mr-Xia-6 are you using a catkin workspace?
No, I do not use the catkin_make,I just make it in desktop.

(omg) yusheng@yusheng:~/desktop/OMG-Planner/orocos_kinematics_dynamics/python_oroco
s_kdl/build$ make
[ 16%] Building CXX object CMakeFiles/PyKDL.dir/PyKDL/frames.cpp.o
In file included from /home/yusheng/桌面/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/PyKDL.h:25:0,
                 from /home/yusheng/桌面/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:27:
/home/yusheng/桌面/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h: In instantiation of ‘void pybind11::cpp_function::initialize(Func&&, Return (*)(Args ...), const Extra& ...) [with Func = pybind11::cpp_function::cpp_function(Return (Class::*)(Arg ...) const, const Extra& ...) [with Return = double; Class = KDL::Vector; Arg = {}; Extra = {pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::arg_v}]::<lambda(const KDL::Vector*)>; Return = double; Args = {const KDL::Vector*}; Extra = {pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::arg_v}]’:
/home/yusheng/桌面/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:100:9:   required from ‘pybind11::cpp_function::cpp_function(Return (Class::*)(Arg ...) const, const Extra& ...) [with Return = double; Class = KDL::Vector; Arg = {}; Extra = {pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::arg_v}]’
/home/yusheng/桌面/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:1267:73:   required from ‘pybind11::class_<type_, options>& pybind11::class_<type_, options>::def(const char*, Func&&, const Extra& ...) [with Func = double (KDL::Vector::*)() const; Extra = {pybind11::arg_v}; type_ = KDL::Vector; options = {}]’
/home/yusheng/桌面/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:92:61:   required from here
/home/yusheng/桌面/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:157:9: error: static assertion failed: The number of argument annotations does not match the number of function arguments
         static_assert(expected_num_args<Extra...>(sizeof...(Args), cast_in::has
         ^
/home/yusheng/桌面/OMG-Planner/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:161:19: error: cannot convert ‘pybind11::cpp_function::initialize(Func&&, Return (*)(Args ...), const Extra& ...) [with Func = pybind11::cpp_function::cpp_function(Return (Class::*)(Arg ...) const, const Extra& ...) [with Return = double; Class = KDL::Vector; Arg = {}; Extra = {pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::arg_v}]::<lambda(const KDL::Vector*)>; Return = double; Args = {const KDL::Vector*}; Extra = {pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::arg_v}]::<lambda(pybind11::detail::function_call&)>’ to ‘pybind11::handle (*)(pybind11::detail::function_call&)’ in assignment
         rec->impl = [](function_call &call) -> handle {
                   ^
CMakeFiles/PyKDL.dir/build.make:94: recipe for target 'CMakeFiles/PyKDL.dir/PyKDL/frames.cpp.o' failed
make[2]: *** [CMakeFiles/PyKDL.dir/PyKDL/frames.cpp.o] Error 1
CMakeFiles/Makefile2:324: recipe for target 'CMakeFiles/PyKDL.dir/all' failed
make[1]: *** [CMakeFiles/PyKDL.dir/all] Error 2
Makefile:159: recipe for target 'all' failed
make: *** [all] Error 2

MatthijsBurgh, 2020-11-09:
Since switching to PyBind version, have you removed your build folder and started completely over?

Are you sure the PyBind submodule is up-to-date?

Mr-Xia-6, 2020-11-09:
> Since switching to PyBind version, have you removed your build folder and started completely over?
> 
> Are you sure the PyBind submodule is up-to-date?

Yes， I started completely over.

’‘’

~/orocos_kinematics_dynamics$ git submodule update --init

Child modules' python_orocos_kdl/pybind11 '(https://github.com/pybind/pybind11.git) to the path' python_orocos_kdl/pybind11 'registration

Being cloned into 'python_orocos_kdl/pybind11'...

remote: Enumerating objects: 13815, done.

remote: Total 13815 (delta 0), reused 0 (delta 0), pack-reused 13815

Receiving object: 100% (13815/13815), 5.33 MiB | 14.00 KiB/s, complete.

Process delta: 100% (9399/9399), complete.

Check the connection... To complete.

Child module path 'python_orocos_kdl/pybind11: check out the' 59 a2ac2745d8a57ac94c6accced73620d59fb844 '

MatthijsBurgh, 2020-11-09:
@meyerj could this be a result of PyKDL finding an older incompatible version of the CPP library?

meyerj, 2020-11-09:
> @meyerj could this be a result of PyKDL finding an older incompatible version of the CPP library?

Yes, possibly. The `python_orocos_kdl` package can only be built with a matching version of `orocos_kdl`, ideally from the exact same commit. Or at least it is not guaranteed to compile or behave as expected, for example if a default argument value in a header would have changed. Normally there should be no ABI changes within a minor release track (1.3, 1.4, ...), but the Python bindings also break for API updates, like adding a new non-virtual method, even if that does not break the ABI.

@Mr-Xia-6 Do you try to build `python_orocos_kdl` against a binary version of KDL installed from [Debian](https://packages.debian.org/source/orocos-kdl), [Ubuntu](https://packages.ubuntu.org/source/orocos-kdl) or [ROS](http://wiki.ros.org/ROS/Installation) repositories? `KDL::ChainJntToJacSolver::updateInternalDataStructures()` was added [5 years ago](https://github.com/orocos/orocos_kinematics_dynamics/commit/2a903664c01141b8c84e54c8306f98830d9b878b), for example, but some of those distributions still deliver [version 1.3](https://github.com/orocos/orocos_kinematics_dynamics/tree/release-1.3) where the method does not existed yet.

Probably we should add a version check in `python_orocos_kdl` and generate a more meaningful error message in CMake already, if the found version of KDL does not match what is expected by `python_orocos_kdl`. For example:
```cmake
project(python_orocos_kdl VERSION 1.4.0)
...
find_package(orocos_kdl REQUIRED ${PROJECT_VERSION_MAJOR}.${PROJECT_VERSION_MINOR} EXACT)

Mr-Xia-6, 2020-11-10:
> python_orocos_kdl

No, I did not try to build python_orocos_kdl against a binary version of KDL installed from Debian, Ubuntu or ROS repositories. I do not know how to use it.

Mr-Xia-6, 2020-11-10:
> Yes, possibly. The `python_orocos_kdl` package can only be built with a matching version of `orocos_kdl`, ideally from the exact same commit. Or at least it is not guaranteed to compile or behave as [...]

No, I did not try to build python_orocos_kdl against a binary version of KDL installed from Debian, Ubuntu or ROS repositories. I do not know how to use it.

MatthijsBurgh, 2020-11-10:
@Mr-Xia-6 How did you install KDL?

MatthijsBurgh, 2020-12-12:
Stale issue, closing

closed 2020-12-12 by MatthijsBurgh