orocos/orocos_kinematics_dynamics #302: KDL::ChainJntToJacSolver should not be used outside of KDL?

traversaro, 2020-11-23:
The `KDL::ChainJntToJacSolver` documentation contains the following phrase: 
~~~
     * @brief  Class to calculate the jacobian of a general
     * KDL::Chain, it is used by other solvers. It should not be used
     * outside of KDL.
~~~
However, the `KDL::ChainJntToJacSolver` is part of the public API and is also exposed in the Python bindings, so it is not clear why "It should not be used outside of KDL.". 

I guess that documentation is just some old leftover, can we remove it?

traversaro, 2020-11-23:
fyi @MatthijsBurgh

MatthijsBurgh, 2020-11-23:
I don't know everything, but I can't come up with any reason why it wouldn't be possible. So just give it a try.

traversaro, 2020-11-23:
> 
> 
> I don't know everything, but I can't come up with any reason why it wouldn't be possible. So just give it a try.

Sorry, I was not clear. `KDL::ChainJntToJacSolver` is perfectly usable from external projects, and it has been like that for a long time, as it also exposed in Python bindings (see https://github.com/orocos/orocos_kinematics_dynamics/blob/7c424ffe77e11b2b7761cd8bb7f91f3abce923df/python_orocos_kdl/tests/kinfamtest.py#L73) and on the top of it more advanced functionality has been added ( https://github.com/orocos/orocos_kinematics_dynamics/pull/64 ). For this reason, I think that this line of documentation is just confusing for users, and probably could be removed. I could go on an do a PR, but I preferred to check with the mantainers if there is any reason for which that line is there.

MatthijsBurgh, 2020-11-23:
Maybe @smits does know why. If he doesn't, we should remove that line.

closed 2020-12-12 by MatthijsBurgh

traversaro, 2020-12-12:
Thanks @MatthijsBurgh !

commit 3c24e768 (2020-12-12): Merge pull request #307 from orocos/MatthijsBurgh-patch-1

fixes #302, Update docstring of ChainJntToJacSolver
commit 9a46b28c (2020-12-12): fixes #302, Update docstring of ChainJntToJacSolver