orocos/orocos_kinematics_dynamics #303: module 'PyKDL' has no attribute 'Tree'

seanng-1, 2020-12-01:
I am trying to use PyKDL alongside the kdl_parser package (https://github.com/ros/kdl_parser) and I encounter this error.

[image]

I installed PyKDL using the install instructions found here https://github.com/orocos/orocos_kinematics_dynamics/blob/master/python_orocos_kdl/INSTALL.md using the catkin method.

I am using the latest version of the commit so I'm not too sure why this isn't working.

MatthijsBurgh, 2020-12-01:
Did you compile using the python3?

seanng-1, 2020-12-06:
I had compiled using catkin but used the manual compilation instructions for Python3 and it seems to be working now.

closed 2020-12-06 by seanng-1