orocos/orocos_kinematics_dynamics #305: ROS2 package failed: orocos_kdl

mkeyno, 2020-12-11:
I've tried yo build the ROS2 package, however, colon build stop on making `orocos_kdl` lib

```
Starting >>> orocos_kdl
[Processing: fastrtps, orocos_kdl]                                             
--- stderr: orocos_kdl                                                         
CMakeFiles/orocos-kdl.dir/segment.cpp.o: file not recognized: File truncated
collect2: error: ld returned 1 exit status
make[2]: *** [src/liborocos-kdl.so.1.4.0] Error 1
make[1]: *** [src/CMakeFiles/orocos-kdl.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< orocos_kdl [39.0s, exited with code 2]
Aborted  <<< fastrtps [6min 33s]  
```

MatthijsBurgh, 2020-12-11:
Can you provide an tree of the RELEVANT folders/files in the build folder?

mkeyno, 2020-12-11:
@MatthijsBurgh  you mean this file 
[t.txt](https://github.com/orocos/orocos_kinematics_dynamics/files/5679309/t.txt)

MatthijsBurgh, 2020-12-11:
Yes, that is what I mean.

The file seems to be available on line 5314. So I don't know what it absolute path it is looking for.

mkeyno, 2020-12-12:
@MatthijsBurgh  there is nothing on `segment.cpp.o` file, I tried to replace it with another file from other VM Linux with fully installed ROS2 , but the error stand still

MatthijsBurgh, 2020-12-12:
What do you mean with "there is nothing on `segment.cpp.o`"?

mkeyno, 2020-12-12:
@MatthijsBurgh ,  it's an empty file
why thre is no segment.cpp file in source files 
there is segment.cpp file in ROS repo but not in ROS2

MatthijsBurgh, 2020-12-12:
Ok, what is the ROS2 repo? I don't know what these people do

MatthijsBurgh, 2020-12-12:
The ROS2 repo is way out of sync with this repo, which is the original one.

I suggest you ask them to get up-to-date with this repo and/or open a PR to get some of their changes in this repo.

I can't help you further for now.

MatthijsBurgh, 2021-01-22:
Stale issue

closed 2021-01-22 by MatthijsBurgh