orocos/orocos_kinematics_dynamics #313: [PyKDL] build error: /usr/bin/sip: No such file or directory

acxz, 2021-01-09:
Hello!

I'm trying to build the library from source however, the build is failing for me.

Build Command:
```
mkdir build
cd build
cmake .. -DPYTHON_VERSION=3
```

Error Message: `make[2]: /usr/bin/sip: No such file or directory`

SysInfo:
OS: Arch Linux
GCC: 10.2.0
CMake: 3.19.2
python: 3.9.1
sip: 6.0.0
orocos_kinematics_dynamics: `v1.4.0` with backported https://github.com/orocos/orocos_kinematics_dynamics/pull/270

Complete Build log:
```
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.9.1", minimum required is "3") 
-- Found PythonLibs: /usr/lib/libpython3.9.so (found suitable version "3.9.1", minimum required is "3.9") 
-- Found SIP version: 4.19.24
-- Configuring done
-- Generating done
-- Build files have been written to: /home/acxz/vcs/git/github/acxz/pkgbuilds/orocos-kdl-python/src/orocos_kinematics_dynamics-1.4.0/python_orocos_kdl
[ 10%] Generating PyKDL/sipPyKDLpart0.cpp, PyKDL/sipPyKDLpart1.cpp, PyKDL/sipPyKDLpart2.cpp, PyKDL/sipPyKDLpart3.cpp, PyKDL/sipPyKDLpart4.cpp, PyKDL/sipPyKDLpart5.cpp, PyKDL/sipPyKDLpart6.cpp, PyKDL/sipPyKDLpart7.cpp

make[2]: /usr/bin/sip: No such file or directory
make[2]: *** [CMakeFiles/python_module_PyKDL.dir/build.make:83: PyKDL/sipPyKDLpart0.cpp] Error 127
make[1]: *** [CMakeFiles/Makefile2:94: CMakeFiles/python_module_PyKDL.dir/all] Error 2
make: *** [Makefile:149: all] Error 2
```

Downstream issue:

acxz, 2021-01-09:
This is most likely due to the release of sip 6.0.0. and I have also seen that we have dropped SIP as a backend and are solely using pybind. I'll try to build with Pybind for 1.4.0 and if that works I'll go ahead and close the issue.

MatthijsBurgh, 2021-01-09:
I have developed the PyKDL based on Pybind. I know I made changes to the cpp library during that process.

I am also pretty sure that `v1.4.0` will not work with Python3 because of the `None` keyword. So you would need to come up with a solution for that.

Also the older SIP based PyKDL is not compatible with SIP 5 and 6. As the sip executable is not available anymore.

acxz, 2021-01-10:
> I am also pretty sure that v1.4.0 will not work with Python3 because of the None keyword. So you would need to come up with a solution for that.

We have been using it (at least compiling it properly) on Arch Linux with ROS Noetic, which is compiled with Python 3.

> ... SIP based PyKDL is not compatible with SIP 5 and 6. As the sip executable is not available anymore.

hmm I feel like we had PyKDL working with SIP 5 as I only saw this issue when I upgraded by SIP to 6.

Anyways, Arch offers the older SIP 4 as well so I was able to fix the Arch Linux package by just building the latest PyKDL release with SIP 4.

Thx for clarifying @MatthijsBurgh 
P.S. would love a new release!

closed 2021-01-10 by acxz