orocos/orocos_kinematics_dynamics #327: In PyKDL, "AttributeError: 'module' object has no attribute 'ChainIdSolver_RNE'"

YummyOu, 2021-03-24:
Now I use Ubuntu 16.04 with ROS kinetic.
I install `orocos_kinematics_dynamics` and `hrl-kdl-indigo-devel`

I would like to compute the torque of each joint by using inverse dynamics. So I use `ChainIdSolver_RNE`, but it doesn't work by PyKDL.
Say `AttributeError: 'module' object has no attribute 'ChainIdSolver_RNE'`

Inverse kinematics is fine for me 
```py
        # IK
        ik_v = PyKDL.ChainIkSolverVel_pinv(self.chain)

        # input
        v_in = PyKDL.Twist(PyKDL.Vector(vel_x, vel_y, vel_z), PyKDL.Vector(0,0,0))
        q_in = PyKDL.JntArray(6)
        for i in range(6):
            q_in[i] = cur_q[i]

        # output
        qdot = PyKDL.JntArray(6)

        res = ik_v.CartToJnt(q_in, v_in, qdot)
```

But dynamics doesn't work.
```py
        self._gravity = PyKDL.Vector(0, 0, -9.81)
        self._wrenches = PyKDL.Wrench()

        gcSolver = PyKDL.ChainIdSolver_RNE(self._chain, self._gravity)
        ret = gcSolver.CartToJnt(q, qd, qdd, self._wrenches, self.tau_gc)

```

MatthijsBurgh, 2021-03-24:
The RNE solver is introduced to PyKDL in e1ca7faf1efa816f2bc12b741d0fda307d60c760. This has not been released in PyKDL 1.4. So the only way to use that feature is to compile from source.

YummyOu, 2021-03-24:
Hi, you mean that I should complie [this version](https://github.com/orocos/orocos_kinematics_dynamics/tree/e1ca7faf1efa816f2bc12b741d0fda307d60c760) to use this solver?

MatthijsBurgh, 2021-03-24:
I recommend you try master first, if that doesn't work, try that specific commit.

YummyOu, 2021-03-24:
> The RNE solver is introduced to PyKDL in [e1ca7fa](https://github.com/orocos/orocos_kinematics_dynamics/commit/e1ca7faf1efa816f2bc12b741d0fda307d60c760). This has not been released in PyKDL 1.4. So the only way to use that feature is to compile from source.

Here 'PyKDL 1.4.' means branch release-1.4 or Tag v1.4.0?

> I recommend you try master first, if that doesn't work, try that specific commit.

What I use now is master branch.

MatthijsBurgh, 2021-03-24:
Yes I mean the branch `release-1.4` That matches the released debians.

Is master working for you?

YummyOu, 2021-03-24:
No, it shows the above error `"AttributeError: 'module' object has no attribute 'ChainIdSolver_RNE'"`
The kinematics functions work, but this dynamics doesn't.

MatthijsBurgh, 2021-03-24:
please print `PyKDL.__file__`, is that where you expect it?

YummyOu, 2021-03-25:
for `PyKDL.__file__`
-> `/opt/ros/kinetic/lib/python2.7/dist-packages/PyKDL.so`

I think that was automatically installed here.

MatthijsBurgh, 2021-03-25:
Yes, so you are not using master, but the released debian, which is matching the `release-1.4` branch.

Please follow the install instructions [here](https://github.com/orocos/orocos_kinematics_dynamics/blob/master/python_orocos_kdl/INSTALL.md) to install the current master version from source.

Please print `PyKDL.__file__` afterwards

YummyOu, 2021-03-25:
I clone the default repository (which is also master branch, right?) and install them by following `INSTALL.md`
 
I test the test file in `python_orocos_kdl` and also print `PyKDL.__file__`

```bash
/opt/ros/kinetic/lib/python2.7/dist-packages/PyKDL.so
testJntSpaceInertiaMatrix (__main__.DynamicsTestFunctions) ... FAIL

======================================================================
FAIL: testJntSpaceInertiaMatrix (__main__.DynamicsTestFunctions)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "../tests/dynamicstest.py", line 39, in testJntSpaceInertiaMatrix
    _ = jm[3, 0]
AssertionError: IndexError not raised

----------------------------------------------------------------------
Ran 1 test in 0.000s

FAILED (failures=1)
```

[image]

MatthijsBurgh, 2021-03-25:
I still doesn't find the correct version. As the debian is in your `$PYTHONPATH` and the manually installed version is probably not.

So there are two options:
- If you are using ROS, please use the catkin method. Make sure that your workspace is overlaying `/opt/ros/$ROS_DISTRO` and not the other way around. That would fix the `$PYTHONPATH` issue directly.
- Add the folder of the installed library to `$PYTHONPATH` (Probably somewhere in `/usr/lib` or `/usr/local/lib`)

YummyOu, 2021-03-30:
Thank you so much! I use the second way to deal with this problem.

My installed `PyKDL.so` is in `'/usr/local/lib/python2.7/dist-packages'`.
In my `$PYTHONPATH`, `'/opt/ros/kinetic/lib/python2.7/dist-packages' `is before than `'/usr/local/lib/python2.7/dist-packages'`. I have no idea to change the order(I have already tried some ways...)

So what I did is to remove the ros path before import PyKDL and insert it after that.
```py
import sys
sys.path.remove('/opt/ros/kinetic/lib/python2.7/dist-packages')
sys.path.insert(0, "/usr/local/lib/python2.7/dist-packages")
import PyKDL
print PyKDL.__file__
sys.path.insert(0,'/opt/ros/kinetic/lib/python2.7/dist-packages')
```

---------------------------------
But, I got `CartToJnt` failed....
Is there anything wrong?

```py

def __init__(self):
       robot = URDF.from_xml_file("/urdf/ur5_ft_robot.urdf")
        tree = kdl_tree_from_urdf_model(robot)

        self._chain = tree.getChain("base", "tool0")

        print self._chain.getNrOfJoints(), "joints"
        print self._chain.getNrOfSegments(), "segments"

        # Param
        self._gravity = PyKDL.Vector(0, 0, -9.81)
        self._wrenches = [PyKDL.Wrench(), PyKDL.Wrench(), PyKDL.Wrench(), PyKDL.Wrench(), PyKDL.Wrench(), PyKDL.Wrench()]

        self.curTau = [0,0,0,0,0,0]
        self.curVel = [0,0,0,0,0,0]
        self.curJoint = [0,0,0,0,0,0]

        self.tau_gc = PyKDL.JntArray(6)

    def computeInvDym(self, q, qd, qdd):

        gcSolver = PyKDL.ChainIdSolver_RNE(self._chain, self._gravity)
        ret = gcSolver.CartToJnt(q, qd, qdd, self._wrenches, self.tau_gc)

        if ret < 0:
            print("CartToJnt Failed")
            return -1

    def move_robot(self):

        q = PyKDL.JntArray(6)
        qd = PyKDL.JntArray(6)
        qdd = PyKDL.JntArray(6)

        t=0

        # cos() curve
        A = 0.4
        T = math.pi
        offset = 0

        print('-------------------- start -----------------------')

        # rate = rospy.Rate(100)
        # 
        while t < T:

            t += 0.0001

            # compute cur_q
            cur_q = A * math.cos(2 * math.pi * t / T) + offset
            # differential q: compute current vel 
            cur_vel = -2 * math.pi / T * A * math.sin(2 * math.pi * t  / T)
            # differential v: compute current acc
            cur_acc = - math.pow(2 * math.pi / T, 2) * A * math.cos(2 * math.pi * t / T)

            q[5] = cur_q
            qd[5] = cur_vel
            qdd[5] = cur_acc

            if self.computeInvDym(q, qd, qdd) == -1:
                break

```

print:
```
Unknown attribute: iyx
Unknown attribute: izx
Unknown attribute: izy
6 joints
8 segments
-------------------- start -----------------------
CartToJnt Failed
```

MatthijsBurgh, 2021-03-30:
@YummyOu Good to hear you got your stuff working.

Please open a new issue for the `CartToJnt` issues.

closed 2021-03-30 by MatthijsBurgh