orocos/orocos_kinematics_dynamics #330: Inconsistent license in orocos_kdl/debian

ralph-lange, 2021-04-29:
According to [orocos_kdl/COPYING](https://github.com/orocos/orocos_kinematics_dynamics/blob/master/orocos_kdl/COPYING) and the headers of all source files (except issue https://github.com/orocos/orocos_kinematics_dynamics/issues/267) the library is licensed under LGPL.

The [copyright](https://github.com/orocos/orocos_kinematics_dynamics/blob/master/orocos_kdl/debian/copyright) file in the [orocos_kdl/debian](https://github.com/orocos/orocos_kinematics_dynamics/blob/master/orocos_kdl/debian/) folder however states that the library is licensed under GPL. Could you clarify this, please?

ralph-lange, 2021-04-29:
Relates to https://github.com/orocos/orocos_kinematics_dynamics/issues/190.

AndyZe, 2021-06-16:
As I understand it, this matters because any project using Orocos/KDL inherits the GPL license. That's quite a few open-source projects!

MatthijsBurgh, 2021-06-16:
@smits @meyerj

smits, 2021-06-17:
> The copyright file in the orocos_kdl/debian folder however states that the library is licensed under GPL. Could you clarify this, please?

That should be updated, the correct license is LGPL.

ralph-lange, 2021-06-17:
Could you please also update orocos_kdl/tests/test-runner.cpp, cf. https://github.com/orocos/orocos_kinematics_dynamics/issues/267?

MatthijsBurgh, 2021-06-18:
@ralph-lange could please apply the desired changes yourself and open a PR?

ralph-lange, 2021-06-18:
Such a contribution to a new repository involves not inconsiderable internal process work for me and would take several days. Moreover, changes to the copyright header should be made by the original author. The simplest thing would be if he simply remove lines 1 through 16 from https://github.com/orocos/orocos_kinematics_dynamics/blob/master/orocos_kdl/tests/test-runner.cpp. Then, the LGPL license from the repository root applies.

MatthijsBurgh, 2021-06-18:
I don't think the changes are that much.

It seems that you already did enough work to find out that the test runner is the only file that needs to be changed. So it wouldn't have been that much extra work. If other changes are needed, please open a PR or drop the line numbers in a new issue if you don't want to edit the lines yourself.

closed 2021-06-18 by MatthijsBurgh

ralph-lange, 2021-06-18:
Thank you very much for resolving the issue in the test folder, @MatthijsBurgh. The issue in the debian folder can be resolved in the same way. If you can remove the copyright file there, the LGPL license from the parent folder will apply.

MatthijsBurgh, 2021-06-19:
Changed the copyright in the debian folder to LGPL.

ralph-lange, 2021-06-21:
Thank you very much for resolving this issue!