orocos/orocos_kinematics_dynamics #332: object has no attribute 'ChainJntToJacDotSolver'  when I use Python

Lejou, 2021-05-26:
ROS: Melodic
IDE: Python

[image]

When I use PyKDL in Python, the strange error has appeared.
Most classes, like ChainJntToJacSolver,ChainDynParam,ChainFkSolverPos_recursive, can be used correctly with atuo-complete in python. But when I want to use class of "ChainJntToJacDotSolver", it is said that:

Traceback (most recent call last):
  File "./dynamic_test.py", line 61, in <module>
    jacDot_kdl = kdl.ChainJntToJacDotSolver(chain)
AttributeError: 'module' object has no attribute 'ChainJntToJacDotSolver'

And I cannot use auto-complete to find this class when I use PyKDL.

Anyone knows the reason? Does this class not exist in PyKDL lib?

MatthijsBurgh, 2021-05-26:
This one is not included in the 1.4 release. Similar to #327. So you will need to build from source if you want to use that solver in python.

closed 2021-05-26 by MatthijsBurgh

Lejou, 2021-05-26:
Appreciate!!