orocos/orocos_kinematics_dynamics #343: under-actuated chains and setWeightTS

BrettRD, 2021-06-25:
I have a 3dof positioning system that only needs to move the end-effector to a point, but doesn't need to care about orientation of the end effector.

From the docs and discussion, `setWeightTS` seems like the correct thing to use, but I can't get any sensible results out of it, nothing ever converges. I'm even feeding a target which is the FK solution of the initial pose.

Are there any working examples of inverse-kinematics on under-actuated (3dof) systems, using `setWeightTS`, that don't resort to virtual joints?  
So far, I've only been able to find complete re-implementations of the IK engine.

guru-florida, 2022-02-14:
FYI I have 3DOF legs on my robots so I had a similar issue.  I've added the necessary extra "virtual" joints at the endpoint so that only endpoint position really matters. Is there a reason why you didn't want to add virtual joints? I also manipulate the joint position in RViz2 using 6DOF manipulators and I have to say having those extra virtual joints makes using the manipulator easier since I can rotate the manipulator into a different rotational axis in order to get the translation plane I want. Or I can lock the virtual joint frame to be parallel to the world frame.

BrettRD, 2022-02-14:
I ended up with a gimbal lock problem within the virtual joints, so I had to extend it to 7dof.
The additional joints add a whole lot of clutter, and they require needlessly complex optimiser to deal with the under-constrained chain.

The docs hint at it being possible to tell the optimiser to ignore specific axes, I'd like to know more about that.