orocos/orocos_kinematics_dynamics #353: Fails to make:  parse error in template argument list

maguilara, 2021-09-06:
Hi, I come from downstream, from the orocos-kdl-python package: https://github.com/acxz/pkgbuilds/issues/145
I get the following error:

```

==> Making package: orocos-kdl-python-git r1039.79adc6f-1 (Mon 06 Sep 2021 04:19:51 PM -05)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Updating orocos_kinematics_dynamics git repo...
Fetching origin
==> Validating source files with sha256sums...
    orocos_kinematics_dynamics ... Skipped
==> Removing existing $srcdir/ directory...
==> Extracting sources...
  -> Creating working copy of orocos_kinematics_dynamics git repo...
Cloning into 'orocos_kinematics_dynamics'...
done.
==> Starting prepare()...
Submodule 'python_orocos_kdl/pybind11' (https://github.com/pybind/pybind11.git) registered for path 'python_orocos_kdl/pybind11'
Cloning into '/home/myusername/.cache/yay/orocos-kdl-python-git/src/orocos_kinematics_dynamics/python_orocos_kdl/pybind11'...
Submodule path 'python_orocos_kdl/pybind11': checked out '787d2c88cafa4d07fb38c9519c485a86323cfcf4'
==> Starting pkgver()...
==> Updated version: orocos-kdl-python-git r1103.bb2493f-1
==> Sources are ready.
==> Making package: orocos-kdl-python-git r1103.bb2493f-1 (Mon 06 Sep 2021 04:19:58 PM -05)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> WARNING: Using existing $srcdir/ tree
==> Starting pkgver()...
==> Starting build()...
-- The C compiler identification is GNU 11.1.0
-- The CXX compiler identification is GNU 11.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.9.6", minimum required is "3") 
-- Found PythonLibs: /usr/lib/libpython3.9.so (found suitable version "3.9.6", minimum required is "3.9") 
-- pybind11 v2.7.1 
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/myusername/.cache/yay/orocos-kdl-python-git/src/orocos_kinematics_dynamics/python_orocos_kdl/build
[ 16%] Building CXX object CMakeFiles/PyKDL.dir/PyKDL/PyKDL.cpp.o
[ 33%] Building CXX object CMakeFiles/PyKDL.dir/PyKDL/frames.cpp.o
[ 50%] Building CXX object CMakeFiles/PyKDL.dir/PyKDL/kinfam.cpp.o
/home/myusername/.cache/yay/orocos-kdl-python-git/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/kinfam.cpp: In function ‘void init_kinfam(pybind11::module&)’:
/home/myusername/.cache/yay/orocos-kdl-python-git/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/kinfam.cpp:81:19: error: parse error in template argument list
   81 |     joint.def(py::init<std::string, Vector, Vector, Joint::JointType, double, double, double, double, double>(),
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/myusername/.cache/yay/orocos-kdl-python-git/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/kinfam.cpp:81:110: error: no matching function for call to ‘init<std::string, <expression error>, <expression error>, KDL::Joint::JointType, double, double, double, double, double>()’
   81 | d::string, Vector, Vector, Joint::JointType, double, double, double, double, double>(),
      | ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~

In file included from /home/myusername/.cache/yay/orocos-kdl-python-git/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/PyKDL.h:26,
                 from /home/myusername/.cache/yay/orocos-kdl-python-git/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/kinfam.cpp:47:
/home/myusername/.cache/yay/orocos-kdl-python-git/src/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:1614:68: note: candidate: ‘template<class ... Args> pybind11::detail::initimpl::constructor<Args ...> pybind11::init()’
 1614 | template <typename... Args> detail::initimpl::constructor<Args...> init() { return {}; }
      |                                                                    ^~~~
/home/myusername/.cache/yay/orocos-kdl-python-git/src/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:1614:68: note:   template argument deduction/substitution failed:
/home/myusername/.cache/yay/orocos-kdl-python-git/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/kinfam.cpp:81:110: error: template argument 2 is invalid
   81 | d::string, Vector, Vector, Joint::JointType, double, double, double, double, double>(),
[... 179 lines cut ...]
In file included from /home/myusername/.cache/yay/orocos-kdl-python-git/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/PyKDL.h:26,
                 from /home/myusername/.cache/yay/orocos-kdl-python-git/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/kinfam.cpp:47:
/home/myusername/.cache/yay/orocos-kdl-python-git/src/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:1621:5: note: candidate: ‘template<class Func, class Ret> Ret pybind11::init(Func&&)’
 1621 | Ret init(Func &&f) { return {std::forward<Func>(f)}; }
      |     ^~~~
/home/myusername/.cache/yay/orocos-kdl-python-git/src/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:1621:5: note:   template argument deduction/substitution failed:
/home/myusername/.cache/yay/orocos-kdl-python-git/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/kinfam.cpp:517:59: error: template argument 2 is invalid
  517 |     chain_id_solver_RNE.def(py::init<const Chain&, Vector>(), py::arg("chain"), py::arg("grav"));
      |                             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
In file included from /home/myusername/.cache/yay/orocos-kdl-python-git/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/PyKDL.h:26,
                 from /home/myusername/.cache/yay/orocos-kdl-python-git/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/kinfam.cpp:47:
/home/myusername/.cache/yay/orocos-kdl-python-git/src/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:1626:5: note: candidate: ‘template<class CFunc, class AFunc, class Ret> Ret pybind11::init(CFunc&&, AFunc&&)’
 1626 | Ret init(CFunc &&c, AFunc &&a) {
      |     ^~~~
/home/myusername/.cache/yay/orocos-kdl-python-git/src/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:1626:5: note:   template argument deduction/substitution failed:
/home/myusername/.cache/yay/orocos-kdl-python-git/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/kinfam.cpp:517:59: error: template argument 2 is invalid
  517 |     chain_id_solver_RNE.def(py::init<const Chain&, Vector>(), py::arg("chain"), py::arg("grav"));
      |                             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
make[2]: *** [CMakeFiles/PyKDL.dir/build.make:104: CMakeFiles/PyKDL.dir/PyKDL/kinfam.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:100: CMakeFiles/PyKDL.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
 -> error making: orocos-kdl-python-git
```

MatthijsBurgh, 2021-09-07:
So what is the command you are running?

And what are your system properties?

maguilara, 2021-09-07:
Sorry! Added to OP some basic info, do you need some more info about my system?

MatthijsBurgh, 2021-09-07:
I think it might be related to you using python 3.9. We would need to check with PyBind11

MatthijsBurgh, 2021-09-07:
@maguilara how is orocos_kdl, the cpp lib, installed?

maguilara, 2021-09-07:
here are the build commands for the orocos-kdl package: https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=orocos-kdl

MatthijsBurgh, 2021-09-07:
But this problem doesn't occur with python 3.9 on ubuntu, so it has to be related to ArchLinux.

I tried the install commands from https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=orocos-kdl-git for the cpp library. That worked fine. Though if you want to run the tests and extend the cmake command with `-DENABLE_TESTS:BOOL=ON` you get conflict with Eigen3.

All issues seems to relate to `Vector`. I don't know what the type of error is. It could be that extending namespacing is the solution. Please give that a try.

MatthijsBurgh, 2021-09-07:
I am pretty sure it has to be related to Vector being resolved to an Eigen Matrix instead of KDL vector

maguilara, 2021-09-08:
> It could be that extending namespacing is the solution. Please give that a try.
How would I do that? Sorry for being clueless

cybaol, 2021-09-08:
I also encountered the same problem. my solution is in kinfam.cpp , please replace all Vector with KDL::Vector !

MatthijsBurgh, 2021-09-08:
The problem was indeed related to Vector being ambiguous, it could also be resolved to an` Eigen::Vector` instead of `KDL::Vector`. Check `namespacing` branch. It solves the problem for me in an Archlinux docker.

maguilara, 2021-09-08:
I'm still getting errors using namespacing, the same ones as before I think. Not sure what could be wrong. I'm using the same PKGBUILD, I've only changed the source to the namespacing branch like this: `source=(git+https://github.com/orocos/orocos_kinematics_dynamics#branch=namespacing)`  I've verified that the files are the new branch indeed.
Not sure if this is the best way to post logs, but I think is better than posting them he
[error.txt](https://github.com/orocos/orocos_kinematics_dynamics/files/7132271/error.txt)

MatthijsBurgh, 2021-09-09:
In the new log:
```
==> Extracting sources...
  -> Creating working copy of orocos_kinematics_dynamics git repo...
Reset branch 'makepkg'
==> Starting prepare()...
```

Old log:
```
==> Extracting sources...
  -> Creating working copy of orocos_kinematics_dynamics git repo...
Cloning into 'orocos_kinematics_dynamics'...
done.
==> Starting prepare()...
```

So it checks out `makepkg` branch if I understand correctly. But that branch doesn't exist in this repo.

Please modify it so you print commit hash. Then we are sure. I am going to merge the PR, so master should be working.

MatthijsBurgh, 2021-09-09:
If your script keeps failing, try doing everything manual including the cloning.

If also the manual steps don't work, drop the versions of all related libraries and tools in here.

maguilara, 2021-09-09:
Yeah, this was an issue on my end. Your commit made changes to orocos-kdl itself on not the python-orocos-kdl, which I tried to compile with the namespacing branch. Changed to orocos-kdl-git, which worked fine, and compiled python-orocos-kdl-git, and it worked fine this time. I think this fixes the issue. I'll close it unless you think differently. 

I'll mention my fix downstream. Thank you :)

closed 2021-09-09 by maguilara