orocos/orocos_kinematics_dynamics #373: Joint class appears to not be thread safe.

Levi-Armstrong, 2021-11-10:
The Joint class has two mutable variables which I believe cause issues if you are using multiple FK solvers each in a different thread created from the same chain.

MatthijsBurgh, 2022-02-16:
You mean this part of the code?

https://github.com/orocos/orocos_kinematics_dynamics/blob/4f705fe9e1533802578b5275ddad02b065b2be4e/orocos_kdl/src/joint.cpp#L79-L85

Levi-Armstrong, 2022-02-16:
That is correct, both q_previous and joint_pose are mutable and were getting manipulated by multiple threads. Would it be an issue to add a mutex or remove the caching?

MatthijsBurgh, 2022-05-24:
@meyerj what is your opinion on this topic? Removing the caching decreases the performance very little IMO. Not having a mutex and lock does have my preference.

Levi-Armstrong, 2022-05-24:
Would thread_local storage for the cache variables solve the issue? I have seen it used but never used it myself so not sure how it works if you are passing around shared pointers to different threads with an object that internally contain a private variable which is thread_local.

closed 2022-06-14 by MatthijsBurgh

meyerj, 2022-06-14:
> @meyerj what is your opinion on this topic? Removing the caching decreases the performance very little IMO. Not having a mutex and lock does have my preference.

Sorry for the late reply: The solution applied in #399 looks good to me.

I am not familiar with the history of KDL either, but likely the caching behavior using mutable variables has been introduced for performance reasons by the original authors based on measurements, for the case that the joint pose is queried multiple times without changes of the joint value, for example when solving forward kinematics in [`ChainFkSolverPos_recursive::JntToCart()`](https://github.com/orocos/orocos_kinematics_dynamics/blob/4f705fe9e1533802578b5275ddad02b065b2be4e/orocos_kdl/src/chainfksolverpos_recursive.cpp). So the patch in #399 may have introduced a performance degradation, but for a very good reason.

If actually needed, it is better to keep the cached joint value and pose with the caller of `KDL::Joint::pose()`, for example as private members of class [`ChainFkSolverPos_recursive`](https://github.com/orocos/orocos_kinematics_dynamics/blob/4f705fe9e1533802578b5275ddad02b065b2be4e/orocos_kdl/src/chainfksolverpos_recursive.cpp).