orocos/orocos_kinematics_dynamics #376: error: package directory 'PyKDL' does not exist

lamuguo, 2022-01-24:
I'm playing clover these days. In running basic_test
(https://github.com/CopterExpress/clover/blob/master/aruco_pose/test/basic.py),
I faced an issue similar to this:
https://answers.ros.org/question/380142/how-to-install-tf2_geometry_msgs-dependency-pykdl/.

Do you know how to solve the issue? Thanks!

MatthijsBurgh, 2022-01-24:
Please let me know how you install both the cpp and the python library.

I can already say, don't use the pypi version.

lamuguo, 2022-01-24:
$ pip install PyKDL
Collecting PyKDL
  Using cached PyKDL-1.4.0.tar.gz (1.5 kB)
    ERROR: Command errored out with exit status 1:
     command: /opt/conda/bin/python3.7 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-71d6y05_/PyKDL/setup.py'"'"'; __file__='"'"'/tmp/pip-install-71d6y05_/PyKDL/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-71d6y05_/PyKDL/pip-egg-info
         cwd: /tmp/pip-install-71d6y05_/PyKDL/
    Complete output (7 lines):
    running egg_info
    creating /tmp/pip-install-71d6y05_/PyKDL/pip-egg-info/PyKDL.egg-info
    writing /tmp/pip-install-71d6y05_/PyKDL/pip-egg-info/PyKDL.egg-info/PKG-INFO
    writing dependency_links to /tmp/pip-install-71d6y05_/PyKDL/pip-egg-info/PyKDL.egg-info/dependency_links.txt
    writing top-level names to /tmp/pip-install-71d6y05_/PyKDL/pip-egg-info/PyKDL.egg-info/top_level.txt
    writing manifest file '/tmp/pip-install-71d6y05_/PyKDL/pip-egg-info/PyKDL.egg-info/SOURCES.txt'
    error: package directory 'PyKDL' does not exist
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

MatthijsBurgh, 2022-01-25:
So like I told before, don't use the pypi version. It is an older version that isn't released by us. So I don't know which python version it supports.

Either use the released debian packages or compile from source.

dimaqq, 2022-01-25:
@MatthijsBurgh there's a process to take control of the pypi project to avoid this mess and/or to publish manylinux binary wheels

zchen24, 2022-01-25:
@MatthijsBurgh @dimaqq I created that Pypi package a long time ago. Happy to add maintainers to that package. 
I do need people's user name. 

[image]

MatthijsBurgh, 2022-01-25:
@dimaqq If we start releasing via pypi, I think it should be done by the maintainers, so we have one group of people that people can contact, so we don't have to send people around to the right person in case of issues.

So are you interested in joining the maintainers?

@zchen24 I will contact you to add me as maintainer. Please don't add any other maintainers for now. So we can first figure out how to do this.

lamuguo, 2022-01-25:
@MatthijsBurgh Nice! In most situation, we have a lot of py env with conda, fix the problem in Pypi would be the best way for developers to use it. Thanks!

dimaqq, 2022-01-26:
I can help building correct manylinux wheels.
I don't understand kimenatics math enough to help in other ways though.

MatthijsBurgh, 2022-01-26:
The current PyPi release shouldn't be used. So by using a debian package or build from source the problem is solved.

closed 2022-01-26 by MatthijsBurgh