orocos/orocos_kinematics_dynamics #377: PyPi releases

MatthijsBurgh, 2022-01-26:
As discussed in https://github.com/orocos/orocos_kinematics_dynamics/issues/376. There seems to be some interest in PyPi releases.

Myself, I don't have any experience in python/PyPi releases. I also have no idea how it works with Python C extensions that require a C/C++ library.

@zchen24 has released one version back in 2018. But wants to add more maintainers which can release newer versions.

@dimaqq Has no knowledge of the kinematics, but has some experience in python releases and would like to join the maintainers for this purpose.

@smits @meyerj What do you think?

dimaqq, 2022-01-26:
> how it works with Python C extensions that require a C/C++ library.

manylinux specifies which libraries are expected in the system: libc, etc.
See https://www.python.org/dev/peps/pep-0599/#the-manylinux2014-policy for full list and exact versions.
Similar list in https://www.python.org/dev/peps/pep-0571/#the-manylinux2010-policy
The newest standard https://www.python.org/dev/peps/pep-0600/ is less clear because it's a meta-standard in a way.

custom libraries, like kdl, need to be bundled into the wheel.
at the moment I'm not clear if the library must be statically linked, or if wheel can include a shared object side-by-side with the Python extension shared object.

My 2c: it may be enough to publish `manylinux2014`, at least at first. That covers Ubuntu 18 and 20.04 and other distros of similar age. Given that ROS2 kinda recommends Ubuntu 20.04 as development base, it's a good fit.

dimaqq, 2022-01-27:
Static linking is fine;
Dynamic linking is OK too, but it falls on the maintainer to ensure that dependent library `soname` won't clash with any other dependent library from any other python package.

luzpaz, 2023-09-24:
Any progress here ?