orocos/orocos_kinematics_dynamics #378: Cannot path test - PyKDL.py

remio87, 2022-01-28:
I'm trying to try the KDL library, and I got the following error when I run PyKDL.py just after build.
```
testJntSpaceInertiaMatrix (dynamicstest.DynamicsTestFunctions) ... python3: /usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:364: Eigen::DenseCoeffsBase<Derived, 1>::Scalar& Eigen::DenseCoeffsBase<Derived, 1>::operator()(Eigen::Index, Eigen::Index) [with Derived = Eigen::Matrix<double, -1, -1>; Eigen::DenseCoeffsBase<Derived, 1>::Scalar = double; Eigen::Index = long int]: Assertion `row >= 0 && row < rows() && col >= 0 && col < cols()' failed.
Aborted
```
Environment:

- Running the test on ROS noetic Docker container running on Ubuntu 20.04 running on Windows 10 (WSL2)
- Cloned the latest version of the master branch
- Build with `catkin build` and `catkin_make` (got same error)
- The following commands have done: `sudo apt-get install libeigen3-dev libcppunit-dev` and `sudo apt-get install python3-psutil python3-future`

Now I don't have any idea to do... Please help me...

remio87, 2022-01-28:
I just cloned and built again, and I got the test passed... Sorry for making a mess!

closed 2022-01-28 by remio87

yaoxt3, 2023-05-25:
Hi, I encountered the same issue, I finished the build, but `python3 tests/PyKDLtest.py` is failed. 
```
testJntSpaceInertiaMatrix (dynamicstest.DynamicsTestFunctions) ... python3: /usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:364: Eigen::DenseCoeffsBase<Derived, 1>::Scalar& Eigen::DenseCoeffsBase<Derived, 1>::operator()(Eigen::Index, Eigen::Index) [with Derived = Eigen::Matrix<double, -1, -1>; Eigen::DenseCoeffsBase<Derived, 1>::Scalar = double; Eigen::Index = long int]: Assertion `row >= 0 && row < rows() && col >= 0 && col < cols()' failed.
Aborted (core dumped)
```
Do you have any ideas to solve this issue? Thank you