orocos/orocos_kinematics_dynamics #382: make error python_orocos_kdl: static assertion failed

zkytony, 2022-02-09:
I was following the instructions [here](https://github.com/orocos/orocos_kinematics_dynamics/blob/master/python_orocos_kdl/INSTALL.md) to build python_orocos_kdl. The make for orocos_kdl C++ finished successfully. 

The error message is: "error: static assertion failed: The number of argument annotations does not match the number of function arguments"

Full make log below:
```
~/repo/robotdev/thirdparty/orocos_kinematics_dynamics/python_orocos_kdl/build $ make
Scanning dependencies of target PyKDL
[ 16%] Building CXX object CMakeFiles/PyKDL.dir/PyKDL/PyKDL.cpp.o
[ 33%] Building CXX object CMakeFiles/PyKDL.dir/PyKDL/frames.cpp.o
In file included from /home/kaiyu/repo/robotdev/thirdparty/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/PyKDL.h:26,
                 from /home/kaiyu/repo/robotdev/thirdparty/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:28:
/home/kaiyu/repo/robotdev/thirdparty/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h: In instantiation of ‘void pybind11::cpp_function::initialize(Func&&, Return (*)(Args ...), const Extra& ...) [with Func = pybind11::cpp_function::cpp_function(Return (Class::*)(Arg ...) const, const Extra& ...) [with Return = double; Class = KDL::Vector; Arg = {}; Extra = {pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::arg_v}]::<lambda(const KDL::Vector*)>; Return = double; Args = {const KDL::Vector*}; Extra = {pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::arg_v}]’:
/home/kaiyu/repo/robotdev/thirdparty/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:126:9:   required from ‘pybind11::cpp_function::cpp_function(Return (Class::*)(Arg ...) const, const Extra& ...) [with Return = double; Class = KDL::Vector; Arg = {}; Extra = {pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::arg_v}]’
/home/kaiyu/repo/robotdev/thirdparty/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:1417:22:   required from ‘pybind11::class_<type_, options>& pybind11::class_<type_, options>::def(const char*, Func&&, const Extra& ...) [with Func = double (KDL::Vector::*)() const; Extra = {pybind11::arg_v}; type_ = KDL::Vector; options = {}]’
/home/kaiyu/repo/robotdev/thirdparty/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:93:61:   required from here
/home/kaiyu/repo/robotdev/thirdparty/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:206:50: error: static assertion failed: The number of argument annotations does not match the number of function arguments
  206 |         static_assert(expected_num_args<Extra...>(sizeof...(Args), cast_in::args_pos >= 0, cast_in::has_kwargs),
      |                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/kaiyu/repo/robotdev/thirdparty/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/attr.h:13,
                 from /home/kaiyu/repo/robotdev/thirdparty/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:13,
                 from /home/kaiyu/repo/robotdev/thirdparty/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/PyKDL.h:26,
                 from /home/kaiyu/repo/robotdev/thirdparty/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:28:
/home/kaiyu/repo/robotdev/thirdparty/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/cast.h:1178:55: error: ‘std::enable_if_t<(! std::is_void<_Res>::value), Return> pybind11::detail::argument_loader<Args>::call(Func&&) && [with Return = double; Guard = pybind11::detail::void_type; Func = pybind11::cpp_function::cpp_function(Return (Class::*)(Arg ...) const, const Extra& ...) [with Return = double; Class = KDL::Vector; Arg = {}; Extra = {pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::arg_v}]::<lambda(const KDL::Vector*)>&; Args = {const KDL::Vector*}; std::enable_if_t<(! std::is_void<_Res>::value), Return> = double]’, declared using local type ‘pybind11::cpp_function::cpp_function(Return (Class::*)(Arg ...) const, const Extra& ...) [with Return = double; Class = KDL::Vector; Arg = {}; Extra = {pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::arg_v}]::<lambda(const KDL::Vector*)>’, is used but never defined [-fpermissive]
 1178 |     enable_if_t<!std::is_void<Return>::value, Return> call(Func &&f) && {
      |                                                       ^~~~
make[2]: *** [CMakeFiles/PyKDL.dir/build.make:76: CMakeFiles/PyKDL.dir/PyKDL/frames.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:297: CMakeFiles/PyKDL.dir/all] Error 2
```
I am using Ubuntu 20.04

MatthijsBurgh, 2022-02-09:
So the issue is line 93 in `PyKDL/frames.cpp`. At which `Vector::Norm` is defined.

I think you forgot to run `sudo make install` on the cpp library. As it probably finds an apt installed version. Which is probably version 1.4. In which the function doesn't have an argument. But it does in master.

Please provide the current commit hash you are on.

zkytony, 2022-02-09:
I was on the most recent commit: 4f705fe9e1533802578b5275ddad02b065b2be4e

I found a work-around for the issue I was facing. I installed `python3-pykdl` instead.

I believe your suggestion is right. I didn't run `sudo make install` for the cpp library. But I have moved on and don't have a way to test it for right now. Will close this issue.

closed 2022-02-09 by zkytony