orocos/orocos_kinematics_dynamics #401: Compiling issue after adding hash function?

martinmatak, 2022-06-10:
I'm following instructions to compile python bindings, but I'm getting an error about the hash function:

```
mmatak@mmatak-dt:~/catkin_ws/src/orocos_kinematics_dynamics$ catkin build python_orocos_kdl 
-----------------------------------------------------------------------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/melodic
Workspace:                   /home/mmatak/catkin_ws
-----------------------------------------------------------------------------------------------------------------------------------------
Build Space:        [exists] /home/mmatak/catkin_ws/build
Devel Space:        [exists] /home/mmatak/catkin_ws/devel
Install Space:      [unused] /home/mmatak/catkin_ws/install
Log Space:          [exists] /home/mmatak/catkin_ws/logs
Source Space:       [exists] /home/mmatak/catkin_ws/src
DESTDIR:            [unused] None
-----------------------------------------------------------------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-----------------------------------------------------------------------------------------------------------------------------------------
Additional CMake Args:       -DPYTHON_INCLUDE_DIR=/usr/include/python3.8 -DPYTHON_INCLUDE_DIR=/usr/include/python3.8 -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.8.so -DPYTHON_VERSION=3.8
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------------------------------------------------------------------------------------
Buildlisted Packages:        None
Skiplisted Packages:         None
-----------------------------------------------------------------------------------------------------------------------------------------
Workspace configuration appears valid.
-----------------------------------------------------------------------------------------------------------------------------------------
[build] Found 9 packages in 0.0 seconds.                                                                                                
[build] Package table is up to date.                                                                                                    
Starting  >>> orocos_kdl                                                                                                                
Finished  <<< orocos_kdl                       [ 7.2 seconds ]                                                                          
Starting  >>> python_orocos_kdl                                                                                                         
________________________________________________________________________________________________________________________________________
Errors     << python_orocos_kdl:make /home/mmatak/catkin_ws/logs/python_orocos_kdl/build.make.008.log                                   
/home/mmatak/catkin_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp: In function ‘void init_frames(pybind11::module&)’:
/home/mmatak/catkin_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:80:33: error: could not convert ‘pybind11::detail::self’ from ‘const pybind11::detail::self_t’ to ‘pybind11::handle’
     vector.def(py::hash(py::self));
                                 ^
/home/mmatak/catkin_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:167:33: error: could not convert ‘pybind11::detail::self’ from ‘const pybind11::detail::self_t’ to ‘pybind11::handle’
     wrench.def(py::hash(py::self));
                                 ^
/home/mmatak/catkin_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:242:32: error: could not convert ‘pybind11::detail::self’ from ‘const pybind11::detail::self_t’ to ‘pybind11::handle’
     twist.def(py::hash(py::self));
                                ^
/home/mmatak/catkin_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:375:35: error: could not convert ‘pybind11::detail::self’ from ‘const pybind11::detail::self_t’ to ‘pybind11::handle’
     rotation.def(py::hash(py::self));
                                   ^
/home/mmatak/catkin_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/framevel.cpp: In function ‘void init_framevel(pybind11::module&)’:
/home/mmatak/catkin_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/framevel.cpp:65:37: error: could not convert ‘pybind11::detail::self’ from ‘const pybind11::detail::self_t’ to ‘pybind11::handle’
     double_vel.def(py::hash(py::self));
                                     ^
/home/mmatak/catkin_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:457:32: error: could not convert ‘pybind11::detail::self’ from ‘const pybind11::detail::self_t’ to ‘pybind11::handle’
     frame.def(py::hash(py::self));
                                ^
/home/mmatak/catkin_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/framevel.cpp:135:37: error: could not convert ‘pybind11::detail::self’ from ‘const pybind11::detail::self_t’ to ‘pybind11::handle’
     vector_vel.def(py::hash(py::self));
                                     ^
/home/mmatak/catkin_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/framevel.cpp:221:36: error: could not convert ‘pybind11::detail::self’ from ‘const pybind11::detail::self_t’ to ‘pybind11::handle’
     twist_vel.def(py::hash(py::self));
                                    ^
/home/mmatak/catkin_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/framevel.cpp:310:39: error: could not convert ‘pybind11::detail::self’ from ‘const pybind11::detail::self_t’ to ‘pybind11::handle’
     rotation_vel.def(py::hash(py::self));
                                       ^
/home/mmatak/catkin_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/framevel.cpp:384:36: error: could not convert ‘pybind11::detail::self’ from ‘const pybind11::detail::self_t’ to ‘pybind11::handle’
     frame_vel.def(py::hash(py::self));
                                    ^
make[2]: *** [CMakeFiles/PyKDL.dir/PyKDL/framevel.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/PyKDL.dir/PyKDL/frames.cpp.o] Error 1
make[1]: *** [CMakeFiles/PyKDL.dir/all] Error 2
make: *** [all] Error 2
cd /home/mmatak/catkin_ws/build/python_orocos_kdl; catkin build --get-env python_orocos_kdl | catkin env -si  /usr/bin/make --jobserver-fds=3,4 -j; cd -

........................................................................................................................................
Failed     << python_orocos_kdl:make           [ Exited with code 2 ]                                                                   
Failed    <<< python_orocos_kdl                [ 6.6 seconds ]                                                                          
[build] Summary: 1 of 2 packages succeeded.                                                                                             
[build]   Ignored:   7 packages were skipped or are skiplisted.                                                                         
[build]   Warnings:  None.                                                                                                              
[build]   Abandoned: None.                                                                                                              
[build]   Failed:    1 packages failed.                                                                                                 
[build] Runtime: 13.9 seconds total.                                

```

MatthijsBurgh, 2022-06-10:
Try to build with `--force-cmake`.

If that doesn't fix the issue, set the cpp version both in `orocos_kdl` and `python_ocoros_kdl` to cpp14. Maybe that will do the job.

Have you tried building the python2 version. As on ubuntu 18.04 the CI succeeds for python2.

martinmatak, 2022-06-10:
Tried adding `--force-cmake` --> same issue
Tied adding `set(CMAKE_CXX_FLAGS "-std=c++14 ${CMAKE_CXX_FLAGS}")` to both `CMakeLists.txt` files --> same issue
Tried combining the two approaches above --> Same issue
Checked out commit `6480921b236baacf282d32857c9a4a1c6f9bd971` --> all good

I don't really want to build it with python 2 because I need to use it with Python3

MatthijsBurgh, 2022-06-11:
I have run a test in GH actions, https://github.com/orocos/orocos_kinematics_dynamics/runs/6841870108, which does succeed. So it should be able to work.

martinmatak, 2022-06-11:
Did you get it to work with `catkin build`?

MatthijsBurgh, 2022-06-11:
I did not test with catkin build in combinatie with 18.04 and python3. Could you check if the plain cmake/make does work for you? (Maybe not the installing, just the cmake and make from https://github.com/orocos/orocos_kinematics_dynamics/blob/master/python_orocos_kdl/INSTALL.md

MannavaVivek, 2022-06-17:
I am facing the same issue with Ubuntu 20.04 and python 3.8 
I have tried all the fixes mentioned here, but nothing is working.

MatthijsBurgh, 2022-06-17:
The CI does test cmake/make on Ubuntu 20.04 with python 3.8. which succeeds.
I am running the same versions on my own machine. I use `catkin build` on it. Which also works. With `catkin build` you need to run it once with `--force-cmake`.

@MannavaVivek please provide more insight in your machine and the steps you took

MannavaVivek, 2022-06-17:
I did use `catkin build` with `--force-cmake` but the results were the same. 
The only thing I could think of is the gcc version (mine is gcc 9.4.0).
I also pulled the code from the commit mentioned above `6480921b236baacf282d32857c9a4a1c6f9bd971`  and was able to build it without issues. In all cases, only python_orocos_kdl had the error, while orocos_kdl built without issues.

MatthijsBurgh, 2022-06-18:
I am not sure what the easiest way to get this information. `--verbose` might do the job. But could you get the found paths of orocos_kdl and pybind during the PyKDL build. When these mismatch we have likely found the cause.

MatthijsBurgh, 2023-01-07:
@MannavaVivek I figured out this is caused by an incompatible pybind version. One installed on your system, not the one in the submodule. If you had responded to my last question about providing the paths of the found libraries we would have found the issue much earlier. Which would have trouble for other people as well.

So next time please respond, when somebody is helping you.