orocos/orocos_kinematics_dynamics #402: KDL::Joint initialization with TransAxis

canalteks, 2022-06-14:
dear developers,
we found following problem (but may not explicit bug) to initialize KDL::Joint with TransAxis.
we hope it could be help.

- Joint::Joint( name, origin, axis, ...) or Joint::Joint( origin, axis, ... ) initialize KDL::Joint with JointType.
- above constructors initialize [joint_pose.M = Rotation::Rot2( axis, offset )](https://github.com/orocos/orocos_kinematics_dynamics/blob/master/orocos_kdl/src/joint.cpp#L54). that means joint axis will be rotated even if JointType == TransAxis.
- but if JointType == TransAxis, we think that joint_pose.M should be Rotation::Identity().
- when computing Joint::pose(), [joint_pose.M variable was not used for TransAxis](https://github.com/orocos/orocos_kinematics_dynamics/blob/master/orocos_kdl/src/joint.cpp#L92) and it does no effect. so that, we cannot see it as an explicit bug.

thanks.

MatthijsBurgh, 2022-06-14:
I didn't write this code, but I think this was just to not have many if/else cases. And as it isn't used, it doesn't matter. But this code is removed in https://github.com/orocos/orocos_kinematics_dynamics/pull/399. So there isn't any question anymore.

canalteks, 2022-06-14:
@MatthijsBurgh , thank you so much.