orocos/orocos_kinematics_dynamics #405: Change License to Apache-2.0

ahoarau, 2022-07-21:
Hi there !

In the context of a private company with proprietary code, having an LGPLv2 can be problematic.
The biggest concern is that it prevents from statically link against the library easily.

I propose to change the license to the more permissive **Apache-2.0**, as you did for all ros2 related stuff 
* https://github.com/orocos/rtt_ros2_integration/blob/master/LICENSE
* https://github.com/orocos/rtt_ros2_common_interfaces/blob/master/LICENSE.

They did the license change in PlotJuggler: https://github.com/facontidavide/PlotJuggler/issues/690
Basically we create an issue and wait for everyone to accept the change.

@MatthijsBurgh What do you think ?

traversaro, 2022-07-21:
Indeed, if that is ok for KDL mantainers it would be great!

MatthijsBurgh, 2022-07-23:
I leave this one to decide to @smits @meyerj.

Also this requires the header of all files to be changed.

smits, 2022-08-09:
@MatthijsBurgh this is ok from my side, you might want to check if there are files that have copyrights not owned by me.

MatthijsBurgh, 2022-08-14:
Yes these files are definitely there.
I am sorry, but I don't feel the responsibility to get consent from all the copyright holders. I think that is up to @ahoarau and @traversaro to do it

ahoarau, 2022-09-01:
ok @MatthijsBurgh I'm closing this to re-open a new one. I'm gonna notify all contributors, not only headers names. I believe this is necessary for legal reasons. 
Also, the MIT license seems more appropriate. That "notice code changes" is cumbersome.

closed 2022-09-01 by ahoarau