orocos/orocos_kinematics_dynamics #410: Change license to MIT

ahoarau, 2022-09-01:
Hello KDL contributors,

This follow the discussion that happened here: https://github.com/orocos/orocos_kinematics_dynamics/issues/405

As discussed with @smits and @MatthijsBurgh, they are both ok to change the license to a more permissive one. `MIT` seems to be the most appropriate (apache-2.0 was considered at first).

To make the change, we need the **consent** of every contributors to make the change from LGPL-v2 to `MIT`.
This will ease proprietary code development, allowing some modifications and static linking.

You can simply respond with "I agree".

Thanks !

NOTE: as a contributor myself, I agree !

@MatthijsBurgh
@smits
@DjoleMNE
@snrkiwi
@eaertbel
@zchen24
@ahoarau
@meyerj
@mvieth
@luzpaz
@jacobperron
@tdelaet
@forrest-rm
@francisco-miguel-almeida
@traversaro
@jbohren
@Tobias-Fischer
@maverick-long
@psoetens
@jf---
@francesco-romano
@jensenb
@ivanpauno
@mikepurvis
@morxa
@JohnSilverFarmer
@dirk-thomas
@aballier
@mag-sruehl
@jbohren-hbr
@martiniil
@sloretz
@spencerjackson
@jspricke
@rethink-kmaroney
@dvanthienen
@mcopejans
@gborghesan
@mvistein
@WouterBancken
@seanyen
@ipa-pgt
@vbargsten
@milutter
@tfoote

morxa, 2022-09-01:
I agree

traversaro, 2022-09-01:
I agree

Tobias-Fischer, 2022-09-01:
I agree

vbargsten, 2022-09-01:
I agree

mag-sruehl, 2022-09-01:
I agree

mvieth, 2022-09-01:
I agree

dvanthienen, 2022-09-01:
I agree

jspricke, 2022-09-01:
I agree

ipa-pgt, 2022-09-01:
I agree

mvistein, 2022-09-01:
I agree

MatthijsBurgh, 2022-09-01:
@ahoarau I am reading into the licenses. I don't see why the LGPL-V2 prevents any larger work (open or closed) to link against this library.

My objection to change to MIT or Apache 2.0 license is that it allows for closed source modifications of this library. This library is open source and I think any modified version should be kept open.

ahoarau, 2022-09-01:
@MatthijsBurgh LGPL-v2 prevents from static linking - at least makes it extremely difficult. "you must provide a way to relink against the library".
Moreover, modifications that are need in some cases when integrating the lib could be difficult to put public for private companies. That could include more minor modifications as simple as building with different flags, fixing some warnings.. all of those are not necessarily useful for the bigger audience + again difficult to put public. 
Apache2 solves point 1, MIT both. 

Many useful libraries like this one have permissive licenses, and still get contributions from people.

jensenb, 2022-09-01:
I agree.

MatthijsBurgh, 2022-09-01:
> I was just lead to believe that you agreed on that in the last issue.

I am sorry I gave you the impression I did. I just wanted to know the opinion of Smits first.

I will come back to you about whether I agree or not.

meyerj, 2022-09-01:
I agree

jspricke, 2022-09-01:
* Antoine Hoarau ***@***.***> [2022-09-01 01:50]:
***@***.*** LGPL-v2 prevents from static linking - at least makes it extremely difficult. "you must provide a way to relink against the library".
> Moreover, modifications that are need in some cases when integrating the lib could be difficult to put public for private companies. That could include more minor modifications as simple as building [...]

FWIW: I I don't see the problem with companies providing the source code 
of those minor modifications. It does not mean that upstream has to 
integrate them. Even more, I think the (L)GPL is especially for cases 
like this that useful changes are contributed back.

>Many useful libraries like this one have permissive licenses, and still get contributions from people.

To me the question is not if there are contributions but rather if those 
how use it successfully contribute back.

Having said that, my contributions here where rather minor so I'm fine 
with whatever is agreed on.

milutter, 2022-09-01:
I agree

maverick-long, 2022-09-01:
I agree

snrkiwi, 2022-09-01:
I agree

> On Aug 31, 2022, at 22:47, Antoine Hoarau ***@***.***> wrote:
> 
> 
> Hello KDL contributors,
> 
> This follow the discussion that happened here: #405 <https://github.com/orocos/orocos_kinematics_dynamics/issues/405>
> As discussed with @smits <https://github.com/smits> and @MatthijsBurgh <https://github.com/MatthijsBurgh>, they are both ok to change the license to a more permissive one. MIT seems to be the most appropriate (apache-2.0 was considered at first).
> 
> To make the change, we need the consent of every contributors to make the change from LGPL-v2 to MIT.
> This will ease proprietary code development, allowing some modifications and static linking.
> 
> You can simply respond with "I agree".
> 
> Thanks !
>

psoetens, 2022-09-01:
Hi Antoine,

The groundwork for KDL was written by Erwin Aertbelien in 2003/2004, while
we developed the Orocos Geometry Library. I did minor contributions and
testing and later,
Ruben took development and maintenance over from Erwin, but both
contributed the largest parts afaikt. Since we used SVN the exact
history/authorship of these days may
be lost but these are the two persons that are instrumental as initial
authors.

I agree for my contributions.

Peter

luzpaz, 2022-09-01:
I agree

eaertbel, 2022-09-01:
Hi Antoine,

I agree to change the license for my contributions to MIT.

Erwin Aertbelien

ivanpauno, 2022-09-01:
I agree

sloretz, 2022-09-01:
I agree

jacobperron, 2022-09-01:
I agree

francesco-romano, 2022-09-01:
I agree

WouterBancken, 2022-09-01:
I agree

tfoote, 2022-09-01:
I agree

forrest-rm, 2022-10-11:
I agree.

jbohren, 2022-10-12:
I agree!

dirk-thomas, 2022-10-12:
I agree.

mikepurvis, 2022-10-12:
I agree

ahoarau, 2022-12-19:
@MatthijsBurgh any news on this ?

MatthijsBurgh, 2022-12-20:
@ahoarau I am sorry, but I can not agree with a change to MIT or apache 2.0.

As statically linking is your biggest issue, I suggest the Mozilla Public License. I would agree with a change to that one. It still requires the user to disclose the source code, when distributing, but it doesn't require compatibility to the original version of the library, So it allows for static linking.
Comparison: https://softwareengineering.stackexchange.com/questions/221365/mozilla-public-license-mpl-2-0-vs-lesser-gnu-general-public-license-lgpl-3-0

ahoarau, 2022-12-20:
@MatthijsBurgh thanks for your feedback. I understand your point of view.

I'm ok with this compromise as well. Probably a simple "patch" on the LGPL could be enough. Something like "LGPL + authorize static linking exception". That might simplify this (long) process.

What do you think ?

ahoarau, 2022-12-22:
> Yes, such a simple patch would also be an option. The question is how we do this correctly?

https://spdx.org/licenses/LGPL-3.0-linking-exception.html

Some like this would be enough I believe. Added to the repo.

zchen24, 2022-12-31:
I agree.

jf---, 2023-01-02:
I agree

mikepurvis, 2023-01-02:
I agree

ahoarau, 2024-02-28:
> @ahoarau you still want to migrate to lgpl 3.0 with the linking exception?
> 
> As mentioned before, I agree with it.

Absolutely. Linking exception, MIT or BSD. Everything that makes life easier, I believe everyone is up for it.

Can you do the commit to upload the linking exception ?

MatthijsBurgh, 2024-02-28:
@ahoarau I can do the commit, can you do the bookkeeping, whether everyone agrees?

ahoarau, 2024-02-28:
> @ahoarau I can do the commit, can you do the bookkeeping, whether everyone agrees? The list above is not complete yet.

Thanks. I believe some accounts are not used anymore, so it's gonna be difficult to keep track of everyone outside Github, but let's give it a last try:
@DjoleMNE @tdelaet @francisco-miguel-almeida @morxa @aballier @jbohren-hbr @martiniil @spencerjackson @rethink-kmaroney @mcopejans @gborghesan @seanyen

Please note that linking exception won't affect fundamentally the meaning of the license, it's a simple made-for-people-building-code extension.

morxa, 2024-02-28:
I agree

dvanthienen, 2024-02-28:
FWDed to alternative email addresses of
@tdelaet @francisco-miguel-almeida  @gborghesan in BCC

gborghesan, 2024-02-29:
Ok for me!

ericriff, 2025-11-28:
Hi all, 

It looks like this is stalled. What's the status of this?

Thanks,

Eric.

mariekecopejans, 2025-11-28:
I agree

DjoleMNE, 2025-12-03:
I agree