orocos/orocos_kinematics_dynamics #428: Python Version Error

ZZWang21, 2022-12-27:
Hi, I installed the PyKDL, but when I import PyKDL, the error came out "ImportError: Python version mismatch: module was compiled for Python 3.8, but the interpreter version is incompatible: 3.7.15 (default, Nov 24 2022, 21:12:53) 
[GCC 11.2.0].". 

I am using UBUNTU 20.04. When i installed, I used: "cmake ../ -DPYTHON_EXECUTABLE:FILEPATH=/home/walter/miniconda3/envs/Robot_ur5e_ws/bin/python"

And it showed: "-- Using CATKIN_DEVEL_PREFIX: /home/walter/orocos_kinematics_dynamics/python_orocos_kdl/build/devel
-- Using CMAKE_PREFIX_PATH: /home/walter/catkin_ws/devel;/opt/ros/noetic
-- This workspace overlays: /home/walter/catkin_ws/devel;/opt/ros/noetic
-- Found PythonInterp: /home/walter/miniconda3/envs/Robot_ur5e_ws/bin/python (found suitable version "3.7.15", minimum required is "3") "

I am not sure what is wrong, please help. 

Thanks!

MatthijsBurgh, 2022-12-27:
Good question.

Pybind11 might be finding its own python version instead of using the already found version.

Try running with the most recent commit, 89e26e0e6acc0f4d8e8a54710232fd2ed80da7b4, on master. As we both swtich the way python is found and we have updated to a newer pybind11 version.

ZZWang21, 2022-12-27:
@MatthijsBurgh  I have already used the latest version: "commit 89e26e0e6acc0f4d8e8a54710232fd2ed80da7b4 (HEAD -> master, origin/master, origin/HEAD)
Author: dependabot[bot] <49699333+dependabot[bot]@users.noreply.github.com>
Date:   Wed Dec 21 23:06:00 2022 +0100"

When I did the cmake, I have already given the conda python 3.7 directory to DPYTHON_EXECUTABLE, but the cmake used the system python 3.8 to compile. 

Stuck here for one day, please help. Thanks.

ZZWang21, 2022-12-27:
@MatthijsBurgh  And, I did the install again, use the latest commit. another error came out "ImportError: liborocos-kdl.so.1.5: cannot open shared object file: No such file or directory"

I am using miniconda3+ python3.7 to use KDL. 

Thanks.

MatthijsBurgh, 2022-12-27:
The log you provided in your first message is definitely from before #425. As that was finding the Python Interpreter. Now the logs should look like this, where it find the python executable.

```bash
Using CATKIN_DEVEL_PREFIX: /home/matthijs/ros/noetic/system/devel/.private/python_orocos_kdl
-- Using CMAKE_PREFIX_PATH: /home/matthijs/ros/noetic/system/devel;/opt/ros/noetic
-- This workspace overlays: /home/matthijs/ros/noetic/system/devel;/opt/ros/noetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: OFF
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- pybind11 not found, building from source
-- pybind11 v2.10.2
-- Configuring done
-- Generating done
-- Build files have been written to: /home/matthijs/ros/noetic/system/build/python_orocos_kdl
```

MatthijsBurgh, 2022-12-27:
> @MatthijsBurgh And, I did the install again, use the latest commit. another error came out "ImportError: liborocos-kdl.so.1.5: cannot open shared object file: No such file or directory"
> 
> I am using miniconda3+ python3.7 to use KDL.
> 
> Thanks.

This sounds very much like a configuration issue. I don't work with conda environments. And especially not in combination with a catkin ws. Just make sure all your environments/ws are sourced correctly and in the correct order. (Don't ask me what the correct order is.) So the resulting environment variables and paths are correct.

MatthijsBurgh, 2023-02-10:
Stale

closed 2023-02-10 by MatthijsBurgh