orocos/orocos_kinematics_dynamics #434: Scaling of RigidBodyInertia values

Afrobeat, 2023-03-12:
Hello,

I realized that when creating a RigidBodyInertia the input values of the COM (h) and the Rotational Inertia (I) are scaled by some kind of value. Here's an example for the values of the COM when initializing a new RigidBodyInertia:
Outside of RigodBodyInertia:
KDL::Vector COM = {-29.110382080078125, -0.00011015097697963938, 93.30975341796875}

Inside RigidBodyInertia:
COM = {-299.4989056680908, -0.001133275989745598, 960.00694734838862}

Same for the Rotational Inertia values. Is there a reason for this? And is there a way to create a new RigidBodyInertia without the scaling? Since I've got the feeling that this scaling might be the reason for huge torque values I get when using the KDL::ChainIdSolver_RNE.

I would be very happy about any help.

MatthijsBurgh, 2023-03-12:
Are you using the constructor correctly?
Please check these [docs](https://tue-robotics.github.io/docs/orocos_kdl/html/classKDL_1_1RigidBodyInertia.html#a1b7ca1ec61df504f4db07cb5f98bdabe)

Also note that `getCOG` returns `h/m` [see here](https://tue-robotics.github.io/docs/orocos_kdl/html/rigidbodyinertia_8hpp_source.html#l00086)

Afrobeat, 2023-03-16:
Thanks for the reply and the links to the documentation.

I actually misunderstood the documentation and did not realize that the `RigidBodyInertia` does not take the `COM` but a vector between a reference point and the COM `(oc)`. 

If I understand the documentation correctly `getCOM()` should return the actual COM. However, it returns `oc`. Is this how it should be?

closed 2023-03-17 by MatthijsBurgh