orocos/orocos_kinematics_dynamics #436: ImportError: dynamic module does not define module export function (PyInit_PyKDL)

nriaziat, 2023-03-23:
I see this was discussed in #96 but I'm still having this issue on the most recent commit. Built both C++ and Python successfully as per the readme, ran ldconfig and still get `"ImportError: dynamic module does not define module export function (PyInit_PyKDL)"`

I passed -DPYTHON-VERSION=3.8 and the appropriate -DPYTHON-EXECUTABLE to cmake. Running the C++ tests with `make check` returned `make: *** No rule to make target 'check'. Stop.`

MatthijsBurgh, 2023-03-23:
Please provide more information about your machine.

Also check you don't have the python 2 version installed.

nriaziat, 2023-03-23:
I’m on Ubuntu 18.04. Python 2 complains about other libraries missing unrelated to KDL, does that indicate it’s not the Python 2 version?

MatthijsBurgh, 2023-03-23:
I have no clue which python version you are running. Please explain all the steps you do and focus on the details.

TONGLoki, 2023-03-24:
Sorry, I have the same problem.
    import PyKDL as p
ImportError: dynamic module does not define module export function (PyInit_PyKDL)
I am using virtual env of python3.6 to run a python 2.7 code. 
But 3.6 has errors:
(pyenv_arl_training) tong@ub2:~/ambf_rl/scripts/dVRK$ python
Python 3.6.9 (default, Feb 28 2023, 09:55:20) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import PyKDL
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: dynamic module does not define module export function (PyInit_PyKDL)
>>> 

python 2.7 works:
tong@ub2:~$ python
Python 2.7.17 (default, Mar  8 2023, 18:40:28) 
[GCC 7.5.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import PyKDL
>>> 
I am confused.

TONGLoki, 2023-03-24:
I have tried the solutions u have posted but they didn't work.

MatthijsBurgh, 2023-03-24:
@TONGLoki I have no clue what you want to do with your 3.6 venv and running 2.7 code.

I can only tell the current master doesn't support python 2.7 anymore.

TONGLoki, 2023-03-24:
I think it is becasue  I use the older package. The training part was using 3.6 env but the libraray was using 2.7?
Traceback (most recent call last):
  File "/home/tong/ambf_rl/scripts/dVRK/PSM_cartesian_herddpg_algorithm.py", line 54, in <module>
    from dVRK.PSM_cartesian_herddpg_env import PSMCartesianHERDDPGEnv
  File "/home/tong/ambf_rl/pyenv_arl_training/lib/python3.6/site-packages/dVRK/__init__.py", line 2, in <module>
    from .PSM_cartesian_env import PSMCartesianEnv, CartesianAction
  File "/home/tong/ambf_rl/pyenv_arl_training/lib/python3.6/site-packages/dVRK/PSM_cartesian_env.py", line 61, in <module>
    from psmFK import compute_FK
  File "/home/tong/ambf/build/devel/lib/python2.7/dist-packages/psmFK/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 2, in <module>
  File "/home/tong/ambf/ambf_controller/dvrk/scripts/utilities.py", line 4, in <module>
    import PyKDL as p
ImportError: dynamic module does not define module export function (PyInit_PyKDL)

TONGLoki, 2023-03-24:
This is someone else's reinforcement learning package, which is also written based on someone else's library, but the library may be very old, so it doesn't support it. But the strange thing is that the original author can use it, and I have such an error report according to the guidance.

nriaziat, 2023-03-24:
@MatthijsBurgh I am using Python 3.8 in Ubuntu 18.04. In order to compile the library I followed these steps:

1. Compile CPP library per INSTALL.MD instructions
2. Go to `python_orocos_kdl/build'
3. Prepare to make with `cmake . -DPYTHON_EXECUTABLE=/usr/bin/python3.8 -DPYTHONVERSION=3.8`. make returns the correct Python Executable with `Found Python: /usr/bin/python3.8`, `Using PYTHON_EXECUTABLE: /usr/bin/python3.8`, `Found PythonInterp: /usr/bin/python3.8`, and `Using Python nosetests: /usr/bin/nosetests-2.7`
4. `make` completes successfully with no errors
5. `sudo make install` completes successfully and installs to `/usr/local/lib/python3/dist-packages/PyKDL.so`
6. LD_LIBRARY_PATH is exported per instructions
7. `sudo ldconfig` runs with no errors
8. `python3.8 ../tests/PyKDLtest.py` returns `ImportError: dynamic module does not define module export function (PyInit_PyKDL)`
9. `python2.7 ../tests/PyKDLtest.py` returns `ImportError: No module named builtins` but manually importing PyKDL in the python 2.7 environment is successful. 

I'm unclear on what more I can do to compile the library for Python 3. Is it because running `python` still links to Python 2.7 by default?

MatthijsBurgh, 2023-03-24:
I think that is happening.

What you could try is `importlib.util.find_spec`. It will return a [spec](https://docs.python.org/3.6/library/importlib.html#importlib.machinery.ModuleSpec) object. Which should tell what it is going to import.

nriaziat, 2023-03-24:
That was very helpful. I didnt realize my ROS installation comes with PyKDL and it's finding the python2.7 origin under the melodic install directory. I was hoping to avoid dealing with ROS for the specific project but it seems I need to make a way around it.

TONGLoki, 2023-03-24:
@nriaziat I tried to bypass 2.7, and successfully installed local 3.6, but I still have this error when running. If you run PyKDL alone there is no error.
```
Python 3.6.9 (default, Mar 10 2023, 16:46:00) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from PyKDL import Vector, Rotation, Frame, dot
>>> 
```
This is the project I need to run. I think I have bypass the 2.7.
```
  File "/home/tong/ambf/ambf_controller/dvrk/scripts/utilities.py", line 1, in <module>
    from PyKDL import Vector, Rotation, Frame, dot
ImportError: dynamic module does not define module export function (PyInit_PyKDL)
```

MatthijsBurgh, 2023-03-24:
@TONGLoki the CWD does influence the import order. But all information can be found in the link provided earlier. Other part of the internet will help you with that.