orocos/orocos_kinematics_dynamics #438: compiling eigen not found

sancelot, 2023-04-19:
Hi,
I am building from source , I setted up all dependencies in /opt/kdl folder
when I run cmake -DCMAKE_PREFIX_PATH=/opt/kdl , orocos does not find my eigen lib.

my eigen lib is available in /opt/kdl/include/eigen3 folder

MatthijsBurgh, 2023-04-19:
Sorry, this sounds like a problem you need to solve yourself. This is related to you doing custom CMake stuff. Which other parts of the internet can help you with.

The only thing I can help you with. When the default way didn't work. We have a custom findEigen. You can provide hints by using `EIGEN3_ROOT` or `EIGEN3_ROOT_DIR`.

Unless you can prove that something is wrong with out CMakelists.txt.

sancelot, 2023-04-19:
Ok, I have been boring  by an eigen package setted up in my ~/.cmake/packages folder... 

This is now working .

closed 2023-04-19 by sancelot