orocos/orocos_kinematics_dynamics #445: Path_RoundedComposite::Clone() doesn't copy F_base_start and F_base_via

DobbyPeng, 2023-08-17:
I'm trying to use Path_RoundedComposite::Clone() to clone a Path_RoundedComposite object.
But I found that Path_RoundedComposite::Clone() doesn't clone F_base_start and F_base_via.
If I call Path_RoundedComposite::finish() on the clone, it doesn't know the last pose.

https://github.com/orocos/orocos_kinematics_dynamics/blob/master/orocos_kdl/src/path_roundedcomposite.cpp#L198
```
Path* Path_RoundedComposite::Clone() {
	return new Path_RoundedComposite(static_cast<Path_Composite*>(comp->Clone()),radius,eqradius,orient->Clone(), true, nrofpoints);
}
```

Should the constructor be modified?

MatthijsBurgh, 2023-08-17:
According to the constructor:
https://github.com/orocos/orocos_kinematics_dynamics/blob/b35de34ab6ca9cb9a776531f519d95d973f83d7f/orocos_kdl/src/path_roundedcomposite.cpp#L55-L59
Everything is taken into account correctly. So the problem is somewhere else.

The variables `F_base_start` and `F_base_via` are marked as cache in the header. But I am  not really sure that is correct.

But this code is really old and I don't have a clue about what is and should be doing. So please make your case why things need to change in a PR.

DobbyPeng, 2023-08-17:
In my case, I want to create a composite path of serval target poses. And in progress, I want to check if poses in the path can be reached by the robot. 

So, I want to clone a path and call the Finish() on the clone. Then get every 1 mm(just for example) of pose for inspection.