orocos/orocos_kinematics_dynamics #447: compilation error

ketchup007, 2023-09-13:
Ubuntu 22.04

I'm trying to compile PyKDL module: got error

In file included from /home/avena/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/PyKDL.h:28,
                 from /home/avena/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:28:
/home/avena/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/operators.h: In instantiation of ‘struct pybind11::detail::op_impl<pybind11::detail::op_hash, pybind11::detail::op_u, KDL::Vector, KDL::Vector, pybind11::detail::undefined_t>’:
/home/avena/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/operators.h:94:24:   required from ‘void pybind11::detail::op_<id, ot, L, R>::execute(Class&, const Extra& ...) const [with Class = pybind11::class_<KDL::Vector>; Extra = {}; pybind11::detail::op_id id = pybind11::detail::op_hash; pybind11::detail::op_type ot = pybind11::detail::op_u; L = pybind11::detail::self_t; R = pybind11::detail::undefined_t]’
/home/avena/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:1600:19:   required from ‘pybind11::class_<type_, options>& pybind11::class_<type_, options>::def(const T&, const Extra& ...) [with T = pybind11::detail::op_<pybind11::detail::op_hash, pybind11::detail::op_u, pybind11::detail::self_t, pybind11::detail::undefined_t>; Extra = {}; typename std::enable_if<T::op_enable_if_hook, int>::type <anonymous> = 0; type_ = KDL::Vector; options = {}]’
/home/avena/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:80:15:   required from here
/home/avena/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/operators.h:187:1: error: use of deleted function ‘std::hash<KDL::Vector>::hash()’
  187 | PYBIND11_UNARY_OPERATOR(hash, hash, std::hash<L>()(l))
      | ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/11/string_view:42,
                 from /usr/include/c++/11/bits/basic_string.h:48,
                 from /usr/include/c++/11/string:55,
                 from /usr/include/c++/11/bits/locale_classes.h:40,
                 from /usr/include/c++/11/bits/ios_base.h:41,
                 from /usr/include/c++/11/ios:42,
                 from /usr/include/c++/11/ostream:38,
                 from /usr/include/c++/11/iostream:39,
                 from /usr/include/kdl/utilities/utility_io.h:27,
                 from /usr/include/kdl/frames_io.hpp:79,
                 from /home/avena/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:27:
/usr/include/c++/11/bits/functional_hash.h:102:12: note: ‘std::hash<KDL::Vector>::hash()’ is implicitly deleted because the default definition would be ill-formed:
  102 |     struct hash : __hash_enum<_Tp>
      |            ^~~~
/usr/include/c++/11/bits/functional_hash.h:102:12: error: no matching function for call to ‘std::__hash_enum<KDL::Vector, false>::__hash_enum()’
/usr/include/c++/11/bits/functional_hash.h:83:7: note: candidate: ‘std::__hash_enum<_Tp, <anonymous> >::__hash_enum(std::__hash_enum<_Tp, <anonymous> >&&) [with _Tp = KDL::Vector; bool <anonymous> = false]’
   83 |       __hash_enum(__hash_enum&&);
      |       ^~~~~~~~~~~
/usr/include/c++/11/bits/functional_hash.h:83:7: note:   candidate expects 1 argument, 0 provided
/usr/include/c++/11/bits/functional_hash.h:102:12: error: ‘std::__hash_enum<_Tp, <anonymous> >::~__hash_enum() [with _Tp = KDL::Vector; bool <anonymous> = false]’ is private within this context
  102 |     struct hash : __hash_enum<_Tp>
      |            ^~~~
/usr/include/c++/11/bits/functional_hash.h:84:7: note: declared private here
   84 |       ~__hash_enum();
      |       ^
In file included from /home/avena/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/PyKDL.h:28,
                 from /home/avena/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:28:
/home/avena/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/operators.h: In instantiation of ‘void pybind11::detail::op_<id, ot, L, R>::execute(Class&, const Extra& ...) const [with Class = pybind11::class_<KDL::Vector>; Extra = {}; pybind11::detail::op_id id = pybind11::detail::op_hash; pybind11::detail::op_type ot = pybind11::detail::op_u; L = pybind11::detail::self_t; R = pybind11::detail::undefined_t]’:
/home/avena/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:1600:19:   required from ‘pybind11::class_<type_, options>& pybind11::class_<type_, options>::def(const T&, const Extra& ...) [with T = pybind11::detail::op_<pybind11::detail::op_hash, pybind11::detail::op_u, pybind11::detail::self_t, pybind11::detail::undefined_t>; Extra = {}; typename std::enable_if<T::op_enable_if_hook, int>::type <anonymous> = 0; type_ = KDL::Vector; options = {}]’
/home/avena/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:80:15:   required from here
/home/avena/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/operators.h:94:28: error: ‘execute’ is not a member of ‘op’ {aka ‘pybind11::detail::op_impl<pybind11::detail::op_hash, pybind11::detail::op_u, KDL::Vector, KDL::Vector, pybind11::detail::undefined_t>’}
   94 |         cl.def(op::name(), &op::execute, is_operator(), extra...);
      |                            ^~~~~~~~~~~~
/home/avena/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/operators.h: In instantiation of ‘struct pybind11::detail::op_impl<pybind11::detail::op_hash, pybind11::detail::op_u, KDL::Wrench, KDL::Wrench, pybind11::detail::undefined_t>’:
/home/avena/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/operators.h:94:24:   required from ‘void pybind11::detail::op_<id, ot, L, R>::execute(Class&, const Extra& ...) const [with Class = pybind11::class_<KDL::Wrench>; Extra = {}; pybind11::detail::op_id id = pybind11::detail::op_hash; pybind11::detail::op_type ot = pybind11::detail::op_u; L = pybind11::detail::self_t; R = pybind11::detail::undefined_t]’
/home/avena/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/pybind11.h:1600:19:   required from ‘pybind11::class_<type_, options>& pybind11::class_<type_, options>::def(const T&, const Extra& ...) [with T = pybind11::detail::op_<pybind11::detail::op_hash, pybind11::detail::op_u, pybind11::detail::self_t, pybind11::detail::undefined_t>; Extra = {}; typename std::enable_if<T::op_enable_if_hook, int>::type <anonymous> = 0; type_ = KDL::Wrench; options = {}]’
/home/avena/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:167:15:   required from here
/home/avena/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/operators.h:187:1: error: use of deleted function ‘std::hash<KDL::Wrench>::hash()’
  187 | PYBIND11_UNARY_OPERATOR(hash, hash, std::hash<L>()(l))
      | ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/11/string_view:42,
                 from /usr/include/c++/11/bits/basic_string.h:48,
                 from /usr/include/c++/11/string:55,
                 from /usr/include/c++/11/bits/locale_classes.h:40,
                 from /usr/include/c++/11/bits/ios_base.h:41,
                 from /usr/include/c++/11/ios:42,
                 from /usr/include/c++/11/ostream:38,
                 from /usr/include/c++/11/iostream:39,
                 from /usr/include/kdl/utilities/utility_io.h:27,
                 from /usr/include/kdl/frames_io.hpp:79,
                 from /home/avena/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:27:
/usr/include/c++/11/bits/functional_hash.h:102:12: note: ‘std::hash<KDL::Wrench>::hash()’ is implicitly deleted because the default definition would be ill-formed:
  102 |     struct hash : __hash_enum<_Tp>
      |            ^~~~
/usr/include/c++/11/bits/functional_hash.h:102:12: error: no matching function for call to ‘std::__hash_enum<KDL::Wrench, false>::__hash_enum()’
/usr/include/c++/11/bits/functional_hash.h:83:7: note: candidate: ‘std::__hash_enum<_Tp, <anonymous> >::__hash_enum(std::__hash_enum<_Tp, <anonymous> >&&) [with _Tp = KDL::Wrench; bool <anonymous> = false]’
   83 |       __hash_enum(__hash_enum&&);
[... 114 lines cut ...]
/usr/include/c++/11/bits/functional_hash.h:102:12: note: ‘std::hash<KDL::Frame>::hash()’ is implicitly deleted because the default definition would be ill-formed:
  102 |     struct hash : __hash_enum<_Tp>
      |            ^~~~
/usr/include/c++/11/bits/functional_hash.h:102:12: error: no matching function for call to ‘std::__hash_enum<KDL::Frame, false>::__hash_enum()’
[... cut ...]
/home/avena/orocos_kinematics_dynamics/python_orocos_kdl/pybind11/include/pybind11/operators.h:94:28: error: ‘execute’ is not a member of ‘op’ {aka ‘pybind11::detail::op_impl<pybind11::detail::op_hash, pybind11::detail::op_u, KDL::Frame, KDL::Frame, pybind11::detail::undefined_t>’}
   94 |         cl.def(op::name(), &op::execute, is_operator(), extra...);
      |                            ^~~~~~~~~~~~
make[2]: *** [CMakeFiles/PyKDL.dir/build.make:90: CMakeFiles/PyKDL.dir/PyKDL/frames.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:100: CMakeFiles/PyKDL.dir/all] Error 2
make: *** [Makefile:136: all] Error 2`

MatthijsBurgh, 2023-09-13:
You really need to provide context.
What python version?
What kdl(cpp) version?
What compiles steps did you do?
What git commit?
etc. etc. etc.

Because in CI it works fine, https://github.com/orocos/orocos_kinematics_dynamics/actions/runs/6174451001, so you are doing something differently....

ketchup007, 2023-09-14:
Ok, it was my problem... I installed ubuntu library dev version and try to compile only python module (with additional bindings).

After compilation and installation cpp library python compilation was correct.

Issue could be closed.

closed 2023-09-14 by ketchup007