orocos/orocos_kinematics_dynamics #464: Failed to build python_orocos_kdl

alemiu, 2024-06-13:
I want  to use tf_conversion in a venv but it returns me the error ModuleNotFoundError: No module named 'PyKDL':

from tf_conversions import transformations
``` 
File "/opt/ros/noetic/lib/python3/dist-packages/tf_conversions/__init__.py", line 30, in <module>
    from .posemath import *
  File "/opt/ros/noetic/lib/python3/dist-packages/tf_conversions/posemath.py", line 34, in <module>
    from PyKDL import *
ModuleNotFoundError: No module named 'PyKDL'
 ```

Then I have followed your instructions to install orocos_kdl and python_orocos_kdl but when doing the catkin make (or catkin build) I get this error:
``` 
=> make -j4 -l4 in '/home/alemiu/DLO_ws/build_isolated/python_orocos_kdl'
Scanning dependencies of target PyKDL
[ 16%] Building CXX object CMakeFiles/PyKDL.dir/PyKDL/PyKDL.cpp.o
[ 33%] Building CXX object CMakeFiles/PyKDL.dir/PyKDL/frames.cpp.o
[ 66%] Building CXX object CMakeFiles/PyKDL.dir/PyKDL/framevel.cpp.o
[ 66%] Building CXX object CMakeFiles/PyKDL.dir/PyKDL/kinfam.cpp.o
/home/alemiu/DLO_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp: In function ‘void init_frames(pybind11::module&)’:
/home/alemiu/DLO_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:80:29: error: could not convert ‘pybind11::detail::self’ from ‘const pybind11::detail::self_t’ to ‘pybind11::handle’
   80 |     vector.def(py::hash(py::self));
      |                         ~~~~^~~~
      |                             |
      |                             const pybind11::detail::self_t
/home/alemiu/DLO_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:167:29: error: could not convert ‘pybind11::detail::self’ from ‘const pybind11::detail::self_t’ to ‘pybind11::handle’
  167 |     wrench.def(py::hash(py::self));
      |                         ~~~~^~~~
      |                             |
      |                             const pybind11::detail::self_t
/home/alemiu/DLO_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:242:28: error: could not convert ‘pybind11::detail::self’ from ‘const pybind11::detail::self_t’ to ‘pybind11::handle’
  242 |     twist.def(py::hash(py::self));
      |                        ~~~~^~~~
      |                            |
      |                            const pybind11::detail::self_t
/home/alemiu/DLO_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/framevel.cpp: In function ‘void init_framevel(pybind11::module&)’:
/home/alemiu/DLO_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/framevel.cpp:65:33: error: could not convert ‘pybind11::detail::self’ from ‘const pybind11::detail::self_t’ to ‘pybind11::handle’
   65 |     double_vel.def(py::hash(py::self));
      |                             ~~~~^~~~
      |                                 |
      |                                 const pybind11::detail::self_t
/home/alemiu/DLO_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:375:31: error: could not convert ‘pybind11::detail::self’ from ‘const pybind11::detail::self_t’ to ‘pybind11::handle’
  375 |     rotation.def(py::hash(py::self));
      |                           ~~~~^~~~
      |                               |
      |                               const pybind11::detail::self_t
/home/alemiu/DLO_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/framevel.cpp:135:33: error: could not convert ‘pybind11::detail::self’ from ‘const pybind11::detail::self_t’ to ‘pybind11::handle’
  135 |     vector_vel.def(py::hash(py::self));
      |                             ~~~~^~~~
      |                                 |
      |                                 const pybind11::detail::self_t
/home/alemiu/DLO_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:457:28: error: could not convert ‘pybind11::detail::self’ from ‘const pybind11::detail::self_t’ to ‘pybind11::handle’
  457 |     frame.def(py::hash(py::self));
      |                        ~~~~^~~~
      |                            |
      |                            const pybind11::detail::self_t
/home/alemiu/DLO_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/framevel.cpp:221:32: error: could not convert ‘pybind11::detail::self’ from ‘const pybind11::detail::self_t’ to ‘pybind11::handle’
  221 |     twist_vel.def(py::hash(py::self));
      |                            ~~~~^~~~
      |                                |
      |                                const pybind11::detail::self_t
/home/alemiu/DLO_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/framevel.cpp:310:35: error: could not convert ‘pybind11::detail::self’ from ‘const pybind11::detail::self_t’ to ‘pybind11::handle’
  310 |     rotation_vel.def(py::hash(py::self));
      |                               ~~~~^~~~
      |                                   |
      |                                   const pybind11::detail::self_t
/home/alemiu/DLO_ws/src/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/framevel.cpp:384:32: error: could not convert ‘pybind11::detail::self’ from ‘const pybind11::detail::self_t’ to ‘pybind11::handle’
  384 |     frame_vel.def(py::hash(py::self));
      |                            ~~~~^~~~
      |                                |
      |                                const pybind11::detail::self_t
[ 83%] Building CXX object CMakeFiles/PyKDL.dir/PyKDL/dynamics.cpp.o
make[2]: *** [CMakeFiles/PyKDL.dir/build.make:76: CMakeFiles/PyKDL.dir/PyKDL/frames.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/PyKDL.dir/build.make:102: CMakeFiles/PyKDL.dir/PyKDL/framevel.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:276: CMakeFiles/PyKDL.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
<== Failed to process package 'python_orocos_kdl': 
  Command '['/home/alemiu/DLO_ws/devel_isolated/orocos_kdl/env.sh', 'make', '-j4', '-l4']' returned non-zero exit status 2.

Reproduce this error by running:
==> cd /home/alemiu/DLO_ws/build_isolated/python_orocos_kdl && /home/alemiu/DLO_ws/devel_isolated/orocos_kdl/env.sh make -j4 -l4

Command failed, exiting.
 ```

And I do not managed to find any other solution on the net. Can someone help me?

MatthijsBurgh, 2024-06-13:
You probably have a system installed version of pybind11 on your machine. See https://github.com/orocos/orocos_kinematics_dynamics/issues/416 and https://github.com/orocos/orocos_kinematics_dynamics/issues/401

You need pybind11>2.6, CMake should find the correct one since https://github.com/orocos/orocos_kinematics_dynamics/pull/430. But I don't know on which commit you are.

MatthijsBurgh, 2024-06-14:
So there is still another version of pybind11 installed on your system. Which is too old. CMake should not use it, as it doesn't match the version requirements. So hopefully that is the case.

When you cloned the repo, did you also initialize the submodule? As we include pybind11 as submodule.

MatthijsBurgh, 2024-06-14:
It doesn't detect pip installed versions of pybind11, see https://github.com/orocos/orocos_kinematics_dynamics/actions/runs/9512645544

It does detect apt installed versions of pybind11, see
https://github.com/orocos/orocos_kinematics_dynamics/actions/runs/9512728029

The last one also shows the CMake version requirement does work.

sara-aldhaheri, 2024-10-25:
Hello,

I'm facing the same issue - I removed all versions of pybind 11 that I have and it still yields the same issue.

TIA

MatthijsBurgh, 2024-10-28:
@sara-aldhaheri please provide a full build log which shows the following lines (or similar):

```
pybind11 not found, building from source
pybind11 v2.13.6
```

sara-aldhaheri, 2024-11-01:
@MatthijsBurgh, I've been trying to install the library again so when I look up pybind now it shows:

```
$ pip show pybind11
Name: pybind11
Version: 2.13.6
```
I've followed the instructions in the INSTALL.md and this is the error log:

```
~/orocos_kinematics_dynamics/python_orocos_kdl/build3$ make 
[ 16%] Building CXX object CMakeFiles/PyKDL.dir/PyKDL/frames.cpp.o
/home/sara/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp: In function ‘void init_frames(pybind11::module&)’:
/home/sara/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:80:29: error: could not convert ‘pybind11::detail::self’ from ‘const pybind11::detail::self_t’ to ‘pybind11::handle’
   80 |     vector.def(py::hash(py::self));
      |                         ~~~~^~~~
      |                             |
      |                             const pybind11::detail::self_t
/home/sara/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:167:29: error: could not convert ‘pybind11::detail::self’ from ‘const pybind11::detail::self_t’ to ‘pybind11::handle’
  167 |     wrench.def(py::hash(py::self));
      |                         ~~~~^~~~
      |                             |
      |                             const pybind11::detail::self_t
/home/sara/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:242:28: error: could not convert ‘pybind11::detail::self’ from ‘const pybind11::detail::self_t’ to ‘pybind11::handle’
  242 |     twist.def(py::hash(py::self));
      |                        ~~~~^~~~
      |                            |
      |                            const pybind11::detail::self_t
/home/sara/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:375:31: error: could not convert ‘pybind11::detail::self’ from ‘const pybind11::detail::self_t’ to ‘pybind11::handle’
  375 |     rotation.def(py::hash(py::self));
      |                           ~~~~^~~~
      |                               |
      |                               const pybind11::detail::self_t
/home/sara/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:457:28: error: could not convert ‘pybind11::detail::self’ from ‘const pybind11::detail::self_t’ to ‘pybind11::handle’
  457 |     frame.def(py::hash(py::self));
      |                        ~~~~^~~~
      |                            |
      |                            const pybind11::detail::self_t
make[2]: *** [CMakeFiles/PyKDL.dir/build.make:76: CMakeFiles/PyKDL.dir/PyKDL/frames.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:276: CMakeFiles/PyKDL.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

```

TIA!

sara-aldhaheri, 2024-11-02:
@MatthijsBurgh , do you mean this log?

```
~/orocos_kinematics_dynamics/python_orocos_kdl/build$ cmake ..
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Python: /usr/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter Development 
-- Using CATKIN_DEVEL_PREFIX: /home/sara/orocos_kinematics_dynamics/python_orocos_kdl/build/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/noetic
-- This workspace overlays: /opt/ros/noetic
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Could NOT find PY_em (missing: PY_EM) 
CMake Error at /opt/ros/noetic/share/catkin/cmake/empy.cmake:30 (message):
  Unable to find either executable 'empy' or Python module 'em'...  try
  installing the package 'python3-empy'
Call Stack (most recent call first):
  /opt/ros/noetic/share/catkin/cmake/all.cmake:164 (include)
  /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:35 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/sara/orocos_kinematics_dynamics/python_orocos_kdl/build/CMakeFiles/CMakeOutput.log".
```
It shows empy cannot be found however it has been installed.

Thanks again!

MatthijsBurgh, 2024-11-02:
Yes this log.

But it doesn't show the pybind lines, as you don't reach that point of the config yet. As it can't find the `empy`. You need to resolve that issue to get more info about the pybind issue.

Make sure the package is installed in the system or user site-packages as you seem to being the system interpreter.