orocos/orocos_kinematics_dynamics #468: Test failure of PyKDL.

threedee003, 2024-08-10:
I installed python_orocos_kdl from the master branch using `catkin_make_isolated` cmd. The installation didnt show any error but the test cases are failing. This is terminal output during the `catkin_make_isolated`. 
```
Base path: /home/tribikram/catkin_ws
Source space: /home/tribikram/catkin_ws/src
Build space: /home/tribikram/catkin_ws/build_isolated
Devel space: /home/tribikram/catkin_ws/devel_isolated
Install space: /home/tribikram/catkin_ws/install_isolated
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~  traversing 2 packages in topological order:
~~  - orocos_kdl (plain cmake)
~~  - python_orocos_kdl
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
The packages or cmake arguments have changed, forcing cmake invocation

==> Processing plain cmake package: 'orocos_kdl'
==> Creating build directory: 'build_isolated/orocos_kdl/devel'
==> cmake /home/tribikram/catkin_ws/src/orocos_kdl -DCMAKE_INSTALL_PREFIX=/home/tribikram/catkin_ws/devel_isolated/orocos_kdl -G Unix Makefiles in '/home/tribikram/catkin_ws/build_isolated/orocos_kdl/devel'
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Orocos KDL version  (1.5.1)
-- Using CATKIN_DEVEL_PREFIX: /home/tribikram/catkin_ws/build_isolated/orocos_kdl/devel/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/noetic
-- This workspace overlays: /opt/ros/noetic
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Found PY_em: /usr/lib/python3/dist-packages/em.py  
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/tribikram/catkin_ws/build_isolated/orocos_kdl/devel/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.10") 
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- Setting build type to 'Release'
-- Performing Test HAVE_STL_CONTAINER_INCOMPLETE_TYPES
-- Performing Test HAVE_STL_CONTAINER_INCOMPLETE_TYPES - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tribikram/catkin_ws/build_isolated/orocos_kdl/devel
==> make -j12 -l12 in '/home/tribikram/catkin_ws/build_isolated/orocos_kdl/devel'
Scanning dependencies of target orocos-kdl
[  4%] Building CXX object src/CMakeFiles/orocos-kdl.dir/chain.cpp.o
[  4%] Building CXX object src/CMakeFiles/orocos-kdl.dir/chaindynparam.cpp.o
[  4%] Building CXX object src/CMakeFiles/orocos-kdl.dir/articulatedbodyinertia.cpp.o
[  5%] Building CXX object src/CMakeFiles/orocos-kdl.dir/chainfdsolver_recursive_newton_euler.cpp.o
[  7%] Building CXX object src/CMakeFiles/orocos-kdl.dir/chainexternalwrenchestimator.cpp.o
[  8%] Building CXX object src/CMakeFiles/orocos-kdl.dir/chainfksolverpos_recursive.cpp.o
[ 13%] Building CXX object src/CMakeFiles/orocos-kdl.dir/chainhdsolver_vereshchagin.cpp.o
[ 13%] Building CXX object src/CMakeFiles/orocos-kdl.dir/chainidsolver_vereshchagin.cpp.o
[ 13%] Building CXX object src/CMakeFiles/orocos-kdl.dir/chainfksolvervel_recursive.cpp.o
[ 14%] Building CXX object src/CMakeFiles/orocos-kdl.dir/chainidsolver_recursive_newton_euler.cpp.o
[ 16%] Building CXX object src/CMakeFiles/orocos-kdl.dir/chainiksolverpos_lma.cpp.o
[... 311 lines cut ...]
======================================================================
ERROR: testJacDot (__main__.KinfamTestFunctions)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "kinfamtest.py", line 58, in setUp
    self.chain.addSegment(Segment(Joint(Joint.Fixed),
AttributeError: type object 'Joint' has no attribute 'Fixed'

======================================================================
ERROR: testTreeGetChainMemLeak (__main__.KinfamTestTree)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "kinfamtest.py", line 331, in setUp
    self.tree.addSegment(Segment(Joint(Joint.Fixed),
AttributeError: type object 'Joint' has no attribute 'Fixed'

----------------------------------------------------------------------
Ran 11 tests in 0.001s

FAILED (errors=11)
```

Is there something wrong that I have done ?

@MatthijsBurgh

MatthijsBurgh, 2024-08-13:
I have not used the old catkin, I have only used [`catkin_tools`](https://github.com/catkin/catkin_tools).

But before using a different catkin, add some print statements to get the path of the PyKDL library. Your python might be importing the incorrect one. Also make sure you resource your workspace after building.

threedee003, 2024-08-14:
Thank you. It's solved. I used `catkin build` and source the workspace.

closed 2024-08-14 by threedee003