orocos/orocos_kinematics_dynamics #478: Interest in a new release?

traversaro, 2024-10-13:
Hello everyone, and thanks a lot for all the work in maintaining KDL!

I am wondering if you are interested in doing a new tag/release of KDL. The last one was some time ago (https://github.com/orocos/orocos_kinematics_dynamics/releases/tag/v1.5.1) and since then several fixes have been added to master, that it would be nice to include in distributions that package the released version of KDL (see https://repology.org/project/orocos-kdl/versions). 

Thanks a lot in advance, and if there is anything I can do to help the release feel free to ask.

snrkiwi, 2024-10-17:
Ditto here too. We are wondering whether this project is actively maintained also? We have some fixes and improvements we'd like to submit.

MatthijsBurgh, 2024-10-18:
I would say it is not being developed, but it is maintained.

@snrkiwi Open a PR I would say.