orocos/orocos_kinematics_dynamics #482: build error

724789975, 2024-11-02:
In file included from /mnt/d/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/PyKDL.h:28,
                 from /mnt/d/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:28:
/usr/include/pybind11/operators.h: In instantiation of ‘struct pybind11::detail::op_impl<pybind11::detail::op_hash, pybind11::detail::op_u, KDL::Vector, KDL::Vector, pybind11::detail::undefined_t>’:
/usr/include/pybind11/operators.h:94:24:   required from ‘void pybind11::detail::op_<id, ot, L, R>::execute(Class&, const Extra& ...) const [with Class = pybind11::class_<KDL::Vector>; Extra = {}; pybind11::detail::op_id id = pybind11::detail::op_hash; pybind11::detail::op_type ot = pybind11::detail::op_u; L = pybind11::detail::self_t; R = pybind11::detail::undefined_t]’
/usr/include/pybind11/pybind11.h:1600:19:   required from ‘pybind11::class_<type_, options>& pybind11::class_<type_, options>::def(const T&, const Extra& ...) [with T = pybind11::detail::op_<pybind11::detail::op_hash, pybind11::detail::op_u, pybind11::detail::self_t, pybind11::detail::undefined_t>; Extra = {}; typename std::enable_if<T::op_enable_if_hook, int>::type <anonymous> = 0; type_ = KDL::Vector; options = {}]’
/mnt/d/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:80:15:   required from here
/usr/include/pybind11/operators.h:187:1: error: use of deleted function ‘std::hash<KDL::Vector>::hash()’
  187 | PYBIND11_UNARY_OPERATOR(hash, hash, std::hash<L>()(l))
      | ^~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/c++/13/string_view:42,
                 from /usr/include/c++/13/bits/basic_string.h:47,
                 from /usr/include/c++/13/string:54,
                 from /usr/include/c++/13/bits/locale_classes.h:40,
                 from /usr/include/c++/13/bits/ios_base.h:41,
                 from /usr/include/c++/13/ios:44,
                 from /usr/include/c++/13/ostream:40,
                 from /usr/include/c++/13/iostream:41,
                 from /usr/include/kdl/utilities/utility_io.h:27,
                 from /usr/include/kdl/frames_io.hpp:79,
                 from /mnt/d/orocos_kinematics_dynamics/python_orocos_kdl/PyKDL/frames.cpp:27:
/usr/include/c++/13/bits/functional_hash.h:102:12: note: ‘std::hash<KDL::Vector>::hash()’ is implicitly deleted because the default definition would be ill-formed:
  102 |     struct hash : __hash_enum<_Tp>

MatthijsBurgh, 2024-11-02:
Please provide more info

ghanta1996, 2025-02-09:
I having the same issue. I'm trying to build ROS2 Foxy from source on MacOS. The compilation fails only at this point. 

[image]

Can anyone help? Thanks

MatthijsBurgh, 2025-02-10:
@ghanta1996 please provide more info than that you are building on MacOS. We don't support MacOS. So when you want any help provide as much information as possible.

Also provide info on which pybind version is being used

ghanta1996, 2025-02-13:
@MatthijsBurgh   Thanks for the response. I'm using pybind 11  (Version: 2.11.1).  The build is mainly failing at the 

 `pybind11_vendor/include/pybind11/operators.h: 187 PYBIND11_UNARY_OPERATOR(hash, hash, std::hash<L>()(l)) `

I was able to build with this commit - `6480921b236baacf282d32857c9a4a1c6f9bd971` which doesn't have hash impl

MatthijsBurgh, 2025-02-14:
I think your issue is the same as https://github.com/orocos/orocos_kinematics_dynamics/issues/447

MatthijsBurgh, 2025-02-14:
Please try my new commit, 05b1676b75494d7cd6f3b66d26ac4e09d4ba6485. It is probably still failing.

MatthijsBurgh, 2025-04-23:
Stale

closed 2025-04-23 by MatthijsBurgh